MarlinFirmware/Marlin #22143: [BUG] (bugfix-2.0.9.x) 'E_AXIS' was not declared in this scope

kpter, 2021-06-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Error while build with  #define EXTRUDERS 0  (CNC config).

> Executing task in folder Marlin-bugfix-2.0.9.x: C:\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.7
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.5
Building in release mode
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
In file included from Marlin\src\HAL\LPC1768\HAL.cpp:26:
Marlin\src\HAL\LPC1768\include/../../../gcode/parser.h: In static member function 'static
float GCodeParser::axis_unit_factor(AxisEnum)':
Marlin\src\HAL\LPC1768\include/../../../gcode/parser.h:314:23: error: 'E_AXIS' was not declared in this scope; did you mean 'C_AXIS'?
  314 |       return (axis >= E_AXIS && volumetric_enabled ? volumetric_unit_factor : linear_unit_factor);
      |                       ^~~~~~
      |                       C_AXIS
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
=============================== [FAILED] Took 8.69 seconds ===============================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:08.687

### Version of Marlin Firmware

Marlin-bugfix-2.0.9.x

### Printer model

CNC

### Electronics

BTT SKR 1.3 board

### Additional information & file uploads

[Configs_n_Diffs.zip](https://github.com/MarlinFirmware/Marlin/files/6658350/Configs_n_Diffs.zip)

slowbro, 2021-06-16:
Looks like it's also doing similar things with `I_AXIS` and the other new axes. I think the `M114_DETAIL` code was missed when 6-axis was added. I also see a lot of references to `E_AXIS` that are not guarded behind a `#if HAS_EXTRUDERS` or the like... I was able to get it to build with these changes:

slowbro/Marlin@75794b3fbb9884c200a02a917c74faa7c7d994a9

I'm not sure if the E_AXIS stuff was all just missed, or like that for some reason? I believe the `EXTRUDERS` option was added a while ago; perhaps setting it to 0 was added more recently.

Let me know if that builds/works for you. I'll see if I can find any more invalid `*_AXIS` references.

DerAndere1, 2021-06-16:
@slowbro there was a recent change in how EXTRUDERS 0 is handled. The goal is that no internal axis is maintained in that case. It looks like the parts you found were missed during the refactoring. I'm looking forward to your next PR.

slowbro, 2021-06-16:
Cool, I wasn't sure if that was the case, or what - I saw the other thread about `EXTRUDERS`, so thanks for clarifying! Would it also be safe to assume that all the `I_AXIS`, `J_AXIS`, and `K_AXIS` should have the same guard conditions?

DerAndere1, 2021-06-16:
thinkyhead did the refactoring related to EXTRUDERS, he might know best. From my perspective, IJK axes should be rather treated like the Y axis inn general. E axis might need guard conditions in other places than those axes.

thisiskeithb, 2021-06-18:
#22163 was merged. Closing.

slowbro, 2021-06-18:
@thisiskeithb I don't think that PR addresses the whole issue; looks like it only changes one file - https://github.com/MarlinFirmware/Marlin/pull/22163/files

I'm working on a PR for the rest, once I have a bit of free time.

kpter, 2021-06-18:
I just tested the latest bugfix-2.0.x.
Merged commits, did not solve the problem.
```
PS D:\CNC\Marlin-Latest\Marlin-bugfix-2.0.9.x> platformio run -e LPC1768
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.7
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.5
Building in release mode
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
In file included from Marlin\src\HAL\LPC1768\HAL.cpp:26:
Marlin\src\HAL\LPC1768\include/../../../gcode/parser.h: In static member function 'static float GCodeParser::axis_unit_factor(AxisEnum)':
Marlin\src\HAL\LPC1768\include/../../../gcode/parser.h:314:23: error: 'E_AXIS' was not declared in this scope; did you mean 'C_AXIS'?
  314 |       return (axis >= E_AXIS && volumetric_enabled ? volumetric_unit_factor : linear_unit_factor);
      |                       ^~~~~~
      |                       C_AXIS
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
============================================ [FAILED] Took 25.37 seconds ============================================
```

ellensp, 2021-06-18:
Give this diff a try 
[diff.txt](https://github.com/MarlinFirmware/Marlin/files/6675720/diff.txt)

ellensp, 2021-06-18:
Give this diff a try 

```DIFF
diff --git a/Marlin/src/gcode/feature/trinamic/M911-M914.cpp b/Marlin/src/gcode/feature/trinamic/M911-M914.cpp
index 747f1c9516..fc0c55155b 100644
--- a/Marlin/src/gcode/feature/trinamic/M911-M914.cpp
+++ b/Marlin/src/gcode/feature/trinamic/M911-M914.cpp
@@ -265,20 +265,22 @@
           TERN_(Z3_HAS_STEALTCHOP, if (index == 0 || index == 3) TMC_SET_PWMTHRS(Z,Z3));
           TERN_(Z4_HAS_STEALTCHOP, if (index == 0 || index == 4) TMC_SET_PWMTHRS(Z,Z4));
           break;
-        case E_AXIS: {
-          #if E_STEPPERS
-            const int8_t target_extruder = get_target_extruder_from_command();
-            if (target_extruder < 0) return;
-            TERN_(E0_HAS_STEALTHCHOP, else if (target_extruder == 0) TMC_SET_PWMTHRS_E(0));
-            TERN_(E1_HAS_STEALTHCHOP, else if (target_extruder == 1) TMC_SET_PWMTHRS_E(1));
-            TERN_(E2_HAS_STEALTHCHOP, else if (target_extruder == 2) TMC_SET_PWMTHRS_E(2));
-            TERN_(E3_HAS_STEALTHCHOP, else if (target_extruder == 3) TMC_SET_PWMTHRS_E(3));
-            TERN_(E4_HAS_STEALTHCHOP, else if (target_extruder == 4) TMC_SET_PWMTHRS_E(4));
-            TERN_(E5_HAS_STEALTHCHOP, else if (target_extruder == 5) TMC_SET_PWMTHRS_E(5));
-            TERN_(E6_HAS_STEALTHCHOP, else if (target_extruder == 6) TMC_SET_PWMTHRS_E(6));
-            TERN_(E7_HAS_STEALTHCHOP, else if (target_extruder == 7) TMC_SET_PWMTHRS_E(7));
-          #endif // E_STEPPERS
-        } break;
+        #if HAS_EXTRUDERS
+          case E_AXIS: {
+            #if E_STEPPERS
+              const int8_t target_extruder = get_target_extruder_from_command();
+              if (target_extruder < 0) return;
+              TERN_(E0_HAS_STEALTHCHOP, else if (target_extruder == 0) TMC_SET_PWMTHRS_E(0));
+              TERN_(E1_HAS_STEALTHCHOP, else if (target_extruder == 1) TMC_SET_PWMTHRS_E(1));
+              TERN_(E2_HAS_STEALTHCHOP, else if (target_extruder == 2) TMC_SET_PWMTHRS_E(2));
+              TERN_(E3_HAS_STEALTHCHOP, else if (target_extruder == 3) TMC_SET_PWMTHRS_E(3));
+              TERN_(E4_HAS_STEALTHCHOP, else if (target_extruder == 4) TMC_SET_PWMTHRS_E(4));
+              TERN_(E5_HAS_STEALTHCHOP, else if (target_extruder == 5) TMC_SET_PWMTHRS_E(5));
+              TERN_(E6_HAS_STEALTHCHOP, else if (target_extruder == 6) TMC_SET_PWMTHRS_E(6));
+              TERN_(E7_HAS_STEALTHCHOP, else if (target_extruder == 7) TMC_SET_PWMTHRS_E(7));
+            #endif // E_STEPPERS
+          } break;
+        #endif
       }
     }
 
diff --git a/Marlin/src/gcode/host/M114.cpp b/Marlin/src/gcode/host/M114.cpp
index 2fdce1edfd..7d69033319 100644
--- a/Marlin/src/gcode/host/M114.cpp
+++ b/Marlin/src/gcode/host/M114.cpp
@@ -216,10 +216,12 @@ void GcodeSuite::M114() {
       report_current_position_detail();
       return;
     }
-    if (parser.seen_test('E')) {
-      SERIAL_ECHOLNPAIR("Count E:", stepper.position(E_AXIS));
-      return;
-    }
+    #if HAS_EXTRUDERS
+      if (parser.seen_test('E')) {
+        SERIAL_ECHOLNPAIR("Count E:", stepper.position(E_AXIS));
+        return;
+      }
+    #endif
   #endif
 
   #if ENABLED(M114_REALTIME)
diff --git a/Marlin/src/gcode/parser.h b/Marlin/src/gcode/parser.h
index 5a1748cc4d..9350fa5584 100644
--- a/Marlin/src/gcode/parser.h
+++ b/Marlin/src/gcode/parser.h
@@ -311,7 +311,7 @@ public:
     }
 
     static inline float axis_unit_factor(const AxisEnum axis) {
-      return (axis >= E_AXIS && volumetric_enabled ? volumetric_unit_factor : linear_unit_factor);
+      return (TERN(HAS_EXTRUDERS, axis >= E_AXIS && volumetric_enabled ? volumetric_unit_factor : linear_unit_factor, linear_unit_factor));
     }
 
     static inline float linear_value_to_mm(const_float_t v)                  { return v * linear_unit_factor; }
diff --git a/Marlin/src/module/motion.cpp b/Marlin/src/module/motion.cpp
index b540c9a938..16f94d4df3 100644
--- a/Marlin/src/module/motion.cpp
+++ b/Marlin/src/module/motion.cpp
@@ -266,7 +266,7 @@ void report_current_position_projected() {
 
     get_cartesian_from_steppers();
     const xyz_pos_t lpos = cartes.asLogical();
-    SERIAL_ECHOPAIR("X:", lpos.x, " Y:", lpos.y, " Z:", lpos.z, " E:", current_position.e);
+    SERIAL_ECHOPAIR("X:", lpos.x, " Y:", lpos.y, " Z:", lpos.z OPTARG(HAS_EXTRUDERS, " E:") OPTARG(HAS_EXTRUDERS, current_position.e));
 
     stepper.report_positions();
     #if IS_SCARA
@@ -1008,7 +1008,9 @@ FORCE_INLINE void segment_idle(millis_t &next_idle_ms) {
       // If the move is very short, check the E move distance
       // No E move either? Game over.
       float cartesian_mm = diff.magnitude();
-      if (UNEAR_ZERO(cartesian_mm)) cartesian_mm = ABS(diff.e);
+      #if HAS_EXTRUDERS
+        if (UNEAR_ZERO(cartesian_mm)) cartesian_mm = ABS(diff.e);
+      #endif
       if (UNEAR_ZERO(cartesian_mm)) return;
 
       // The length divided by the segment size

```

slowbro, 2021-06-21:
#22176 was just merged that should fix this.

PR #22176 (2021-06-21): Fix IJK axis references and E stepper indices
### Description

Fixes some references to `E_AXIS` that break compilation with `#define EXTRUDERS 0`, as well as the new linear axes i/j/k.

Also includes some cleanup/standardization for multiple E steppers in M569, M906, and M911-914.

### Requirements

EXTRUDERS 0 for the E_AXIS stuff, LINEAR_AXES < 4.

### Benefits

Fixes some edge cases in compilation.

### Configurations

See #22143

### Related Issues

Fixes #22143
closed 2021-06-22 by thisiskeithb