MarlinFirmware/Marlin #22145: [BUG] Multiple Extruder INSANITY

rrroonn, 2021-06-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have two extruders and two hotends on a machine I am developing.  I came across weirdness when testing the extruders as the second extruder would extrude four times as much as the first.  I did a fair bit of debugging (including trying the latest bugfix)  and eventually noticed that I has not defined a driver for this channel.

IE, I had

`#define E0_DRIVER_TYPE TMC2209
`
but

`//#define E1_DRIVER_TYPE A4988
`
Despite this, the axis did work but not correctly.  There also seemed to be some corruption when using M92.

Obviously, I had incompletely configured Marlin, but it would be good to catch this via an additional SANITY check.

thanks,

Ron

### Bug Timeline

no idea

### Expected behavior

I expect that (a) the second extruder should not work if a driver is not configured, and
(b) a compile time warning should be emitted if functions that require a dual extruder are enabled but an additional driver is not enabled.

### Steps to Reproduce

1. configure for dual nozzles
2. do not specify a driver type for the second extruder
3. check steps per mm for E1 vs E0

### Version of Marlin Firmware

current bugfix downloaded today

rrroonn, 2021-06-16:
SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Jun 16 2021 15:07:01) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:NEWBOT-1 EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0

ellensp, 2021-06-16:
All drivers are presumed to be A4988 if not explicitly set. Thus you will not get warnings that driver is not set, ever.

closed 2021-06-16 by ellensp

rrroonn, 2021-06-16:
> All drivers are presumed to be A4988 if not explicitly set. Thus you will not get warnings that driver is not set, ever.

Ok, I am sorry that you seem to take offence.  Did you read where M92 corruption occurred?  Anyway, I will not report bugs in future.