MarlinFirmware/Marlin #22146: MKS Robin E3 Compile error "variant.h: No such file or directory"

McTheodoren, 2021-06-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Just installed an MKS robin e3 and was trying to update it to marlin 2.0.9 but i get an error

[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = mks_robin_e3
include_dir = Marlin

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MKS_ROBIN_E3
#endif

works on 2.8.2 but when i updated to 2.0.9 it gives the error

### Bug Timeline

from 2.8.2 to 2.0.9

### Expected behavior

successfull compilaton

### Actual behavior

* CLI  > platformio lib search "header:variant.h"
* Web  > https://platformio.org/lib/search?query=header:variant.h
*
*****************************************************************

    4 | #include "variant.h"
      |          ^~~~~~~~~~~
compilation terminated.
In file included from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal.h:30,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include/stm32f1xx.h:200,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/stm32_def.h:28,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/clock.h:43,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/wiring_time.h:23,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/wiring.h:38,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/Arduino.h:36,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\IPAddress.cpp:20:
C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\STM32F1xx/stm32f1xx_hal_conf.h:4:10: fatal error: variant.h: No such file or directory

*****************************************************************
* Looking for variant.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:variant.h"
* Web  > https://platformio.org/lib/search?query=header:variant.h
*
*****************************************************************

    4 | #include "variant.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_e3\FrameworkArduino\HardwareTimer.cpp.o] Error 1
*** [.pio\build\mks_robin_e3\FrameworkArduino\IPAddress.cpp.o] Error 1
In file included from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal.h:30,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include/stm32f1xx.h:200,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/stm32_def.h:28,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/clock.h:43,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\wiring_time.h:23,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\wiring.h:38,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\Arduino.h:36,
                 from C:\Users\marqu\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\HardwareSerial.cpp:26:
[... 76 lines cut ...]
    4 | #include "variant.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_e3\FrameworkArduino\Tone.cpp.o] Error 1

### Steps to Reproduce

1. download latest marlin install
2. change the config file and the pio file to mks robin e3

### Version of Marlin Firmware

2.0.9

### Printer model

anet AM8 with MKS robin e3

### Electronics

MKS robin e3

### Additional information & file uploads

[MArlin.zip](https://github.com/MarlinFirmware/Marlin/files/6663015/MArlin.zip)

thisiskeithb, 2021-06-16:
Confirmed. The usual trick of deleting the hidden `.pio` folder in Marlin as well as `.platformio` in my user folder doesn't resolve it.

It's not ideal, but you can compile with `mks_robin_e3_maple` this can be tracked down.

spooky79, 2021-06-16:
I have the same error :/

thisiskeithb, 2021-06-16:
> I have the same error :/

Thanks for the report, but the bug has already been marked as confirmed.

ellensp, 2021-06-16:
I cannot replicate this issue... so whats different?  (Linux probably) 
I tested bugfix, change MOTHERBOARD and SERIAL_PORT and build env.
@thisiskeithb can you compare version numbers etc with my log?
[build_log.zip](https://github.com/MarlinFirmware/Marlin/files/6666265/build_log.zip)

thisiskeithb, 2021-06-16:
This was tested with VSCode 1.57.0 + PIO Core 5.1.1 / Home 3.3.4 under macOS Catalina 10.15.7.

<details><summary>Full output:</summary>

```prolog
Processing mks_robin_e3 (platform: ststm32@~12.1; board: genericSTM32F103RC; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/mks_robin_e3/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/Print.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/RingBuffer.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/mks_robin_e3/FrameworkArduino/USBSerial.cpp.o
In file included from /Users/keith/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal.h:30,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f1xx.h:200,
Compiling .pio/build/mks_robin_e3/FrameworkArduino/VirtIOSerial.cpp.o
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/stm32_def.h:28,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/clock.h:43,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring_time.h:23,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:38,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/HardwareSerial.cpp:26:
/Users/keith/.platformio/packages/framework-arduinoststm32In file included from /Users/keith/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal.h:30,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f1xx.h:200,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/stm32_def.h:28,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/clock.h:43,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring_time.h:23,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:38,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from /Users/keith/.platformio/packages/framework-arduinoststm32/cores/arduino/HardwareTimer.cpp:26:
/Users/keith/.platformio/packages/framework-arduinoststm32/system/STM32F1xx/stm32f1xx_hal_conf.h:4:10: fatal error: variant.h: No such file or directory

*****************************************************************
* Looking for variant.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:variant.h"
* Web  > https://platformio.org/lib/search?query=header:variant.h
*
*****************************************************************

    4 | #include "variant.h"
[... 200 lines cut ...]
*
*****************************************************************

    4 | #include "variant.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio/build/mks_robin_e3/FrameworkArduino/Print.cpp.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/Stream.cpp.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/Tone.cpp.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/WSerial.cpp.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/board.c.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/WInterrupts.cpp.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/pins_arduino.c.o] Error 1
*** [.pio/build/mks_robin_e3/FrameworkArduino/main.cpp.o] Error 1
======================================================================================================================== [FAILED] Took 5.91 seconds ========================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_robin_e3   FAILED    00:00:05.909
=================================================================================================================== 1 failed, 0 succeeded in 00:00:05.909 ===================================================================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
</details>

rhapsodyv, 2021-06-16:
Fixed here: #22149

rhapsodyv, 2021-06-16:
> I cannot replicate this issue... so whats different? (me running Linux probably)
> I tested bugfix, change MOTHERBOARD and SERIAL_PORT and build env.
> @thisiskeithb can you compare version numbers etc with my log?
> [build_log.zip](https://github.com/MarlinFirmware/Marlin/files/6666265/build_log.zip)

The script that copies the marlin variant to the framework folder wasn’t running. If you already have the variant copied, you get no error. If you start from zero and never build any F103 before, this env will fail because the copy will not be done.

PR #22149 (2021-06-17): 🍬Fix #22146 - mks_robin_e3 build issue
Fix #22146 - mks_robin_e3 build issue (pointing to the wrong base scripts).

The build fail because the right scripts aren't executed.
closed 2021-06-17 by thinkyhead