MarlinFirmware/Marlin #22148: [BUG] 2.0.9 won't compile for GTR 1.0 w/ MAX31865

truglodite, 2021-06-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling fails using bugfix configured for the GTR 1.0 board with `TEMP_SENSOR_0 -5`. This happens with or without a 6675 cs pin defined.

### Bug Timeline

2.0.9

### Expected behavior

Compile, work, and move on

### Actual behavior

Won't compile, and the error messages appear to point a finger at the recent seemingly significant changes to formatting of macros:

```
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\temperature.cpp:30:
Marlin\src\module\../inc/../core/macros.h:193:48: error: pasting "ENA_" and "(" does not give a valid preprocessing token
  193 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                                ^~~~
Marlin\src\module\../inc/../core/macros.h:184:22: note: in definition of macro '_CAT'
  184 | #define _CAT(a,V...) a##V
      |                      ^
Marlin\src\module\../inc/../core/macros.h:193:44: note: in expansion of macro 'CAT'
  193 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                            ^~~
Marlin\src\module\../inc/../core/macros.h:202:35: note: in expansion of macro '_ENA_1'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                                   ^~~~~~
Marlin\src\module\temperature.cpp:2093:5: note: in expansion of macro 'TERN_'
 2093 |     TERN_(TEMP_SENSOR_IS_MAX(0, MAX31865), max31865_0.begin(MAX31865_2WIRE)); // MAX31865_2WIRE, MAX31865_3WIRE, MAX31865_4WIRE
      |     ^~~~~
Marlin\src\module\../inc/../core/macros.h:193:48: error: pasting "ENA_" and "(" does not give a valid preprocessing token
  193 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                                ^~~~
Marlin\src\module\../inc/../core/macros.h:184:22: note: in definition of macro '_CAT'
  184 | #define _CAT(a,V...) a##V
      |                      ^
Marlin\src\module\../inc/../core/macros.h:193:44: note: in expansion of macro 'CAT'
  193 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                            ^~~
Marlin\src\module\../inc/../core/macros.h:202:35: note: in expansion of macro '_ENA_1'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                                   ^~~~~~
Marlin\src\module\temperature.cpp:2094:5: note: in expansion of macro 'TERN_'
 2094 |     TERN_(TEMP_SENSOR_IS_MAX(1, MAX31865), max31865_1.begin(MAX31865_2WIRE));
      |     ^~~~~
Archiving .pio\build\BIGTREE_GTR_V1_0\libc13\libSPI.a
Compiling .pio\build\BIGTREE_GTR_V1_0\lib724\Adafruit MAX31865 library\Adafruit_MAX31865.cpp.o
Indexing .pio\build\BIGTREE_GTR_V1_0\libc13\libSPI.a
Marlin\src\module\temperature.cpp: In member function 'void Temperature::init()':
Marlin\src\module\temperature.cpp:2094:44: error: 'max31865_1' was not declared in this scope; did you mean 'max31865_0'?
 2094 |     TERN_(TEMP_SENSOR_IS_MAX(1, MAX31865), max31865_1.begin(MAX31865_2WIRE));
      |                                            ^~~~~~~~~~
Marlin\src\module\../inc/../core/macros.h:558:26: note: in definition of macro 'THIRD'
  558 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\module\../inc/../core/macros.h:204:29: note: in expansion of macro '___TERN'
  204 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:203:29: note: in expansion of macro '__TERN'
  203 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:202:29: note: in expansion of macro '_TERN'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\module\temperature.cpp:2094:5: note: in expansion of macro 'TERN_'
 2094 |     TERN_(TEMP_SENSOR_IS_MAX(1, MAX31865), max31865_1.begin(MAX31865_2WIRE));
      |     ^~~~~
Compiling .pio\build\BIGTREE_GTR_V1_0\lib8b2\Servo\stm32\Servo.cpp.o
*** [.pio\build\BIGTREE_GTR_V1_0\src\src\module\temperature.cpp.o] Error 1
========================= [FAILED] Took 64.48 seconds =========================
 
Environment       Status    Duration
----------------  --------  ------------
BIGTREE_GTR_V1_0  FAILED    00:01:04.479
==================== 1 failed, 0 succeeded in 00:01:04.479 ====================
 
```

### Steps to Reproduce

1. Get the latest bugfix
2. Add to configuration.h:
`#define TEMP_SENSOR_0 -5`
3. Optional... you can try and configure the CS pin, but it won't change anything.
4. Compile, and sigh

### Version of Marlin Firmware

Can't do M115... so whatever was up right before writing this.

### Printer model

A8... or used to be an fine upstanding A8

### Electronics

GTR 1.0 w/ 5x TMC2209

### Add-ons

Adafruit MAX31865 RTD breakout board

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

I did not include my configs here, because there really weren't any changes other than defining the T0 sensor type. Again, I tried playing with the pins file in case this was related to not having a CS pin formally defined. That did not help at all.

On a side note, the same hardware with 2.0.8.2 is having issues with USB comms when using the -5 sensor. Not sure if that is related to my sd card config, or maybe has something to do with this bug.

ellensp, 2021-06-16:
Please attach your full configuration files.  (and any pin changes)

truglodite, 2021-06-17:
Here are the configs:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6666525/configs.zip)

There are less than a handful of lines changed; marked them all with "trug" in the comments. Just to be clear, it compilation fails both with and without the change to the pins file.

ellensp, 2021-06-17:
This is odd...

In Marlin\src\module\temperature.cpp the first line that fails is TERN_(TEMP_SENSOR_IS_MAX(0, MAX31865), max31865_0.begin(MAX31865_2WIRE));

if you dig through the macros in this case the line  can be simplified to
TERN_((_ENA_1(1)), max31865_0.begin(MAX31865_2WIRE));
and still fails with "Marlin/src/module/temperature.cpp:2102:48: error: pasting "ENA_" and "(" does not give a valid preprocessing token"  as expected.

It should convert this from  (_ENA_1(1))  -->  _ISENA(CAT(IS,CAT(ENA, 1)))  --> _ISENA(_ISENA_1)
but it seems like it is not generating _ISENA(CAT(IS,CAT(ENA, 1)))  correctly...

This macro defined as #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
It seems like parameter O it is not being passed correctly.

if if I wrap that TERN_ with an override, forcing parameter O to be 1 always eg
#undef _ENA_1
#define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, 1)))
TERN_((_ENA_1(1)), max31865_0.begin(MAX31865_2WIRE));
#undef _ENA_1
#define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
The line compiles correctly.

This is very strange...  perhaps some else can clarify what is going on...

slowbro, 2021-06-19:
This is fixed in #22173 (awaiting merge). The `TERN_()` macro can't be used with non-options, which I was not aware of when writing that!

slowbro, 2021-06-19:
#22173 was merged so this should be fixed!

thisiskeithb, 2021-06-19:
> #22173 was merged so this should be fixed!

I had to enable `USE_YMIN_PLUG` in OP's config, but it now builds.

closed 2021-06-19 by thisiskeithb