MarlinFirmware/Marlin #22153: [BUG] Ender 3 V2 - Error:MINTEMP triggered, system stopped! Heater_ID: 0 - broken from 2021-06-10 onwards

CRCinAU, 2021-06-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

For builds on 10th of this month and later, any command to heat the hotend will throw a MINTEMP error as follows:

```
Recv:  T:20.14 /0.00 B:20.27 /0.00 @:0 B@:0
Send: M104 S200
Recv: ok P15 B3
Recv: Error:MINTEMP triggered, system stopped! Heater_ID: 0
Recv: echo:Err: MINTEMP
```

Reverting back to a build ~10AM AEST on the 9th and this problem does not appear.

The heat command can come from USB UART, or from the devices screen to reproduce the error.

### Bug Timeline

2021-06-10

### Expected behavior

After a target temp being set, the hotend should heat to that.

### Actual behavior

```
Error:MINTEMP triggered, system stopped! Heater_ID: 0
```

Unsure if this also affects v4.2.7 mainboards.

### Steps to Reproduce

Ender 3 V2 build after around 2021-06-10 onwards.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 V2

### Electronics

Creality v4.2.2 mainboard

### Additional information & file uploads

Initial suspect of this behaviour is this commit:
https://github.com/MarlinFirmware/Marlin/commit/77496c8235dab5182312652b4dc0b6d0f182cf30

However I can't confirm that at this point.

CRCinAU, 2021-06-17:
When performing a PID Tune on the 20210610 build, you can see things heat up ok, the thermisor registers the correct temps, and the process stops. Once the hotend is given a target temp, the MINTEMP error triggers... As logged in Octoprint:
[image]

Doing just a normal heat of the hotend with 20210609 build works fine:
[image]

ellensp, 2021-06-17:
I can confirm I replicated the issue using a provided firmware.bin, so far trigger is elusive

CRCinAU, 2021-06-17:
It sometimes seems hard to start the triggering off - but It does seem that doing:

* Restore Settings
* Store Setings
* Powercycle Printer
* Attempt to heat hotend

Can make this more likely to occur... Once its happened once, it seems to happen every time...

CRCinAU, 2021-06-19:
As a note to make this reproducible... If I flash a new firmware, then Restore Settings -> Store Settings then heat the hotend, it works. If I power the printer off, then on again and try to heat the hotend, the failure is immediate.

Further Restore Settings -> Store Settings attempts do not restore the ability to heat the hotend and still produce the MINTEMP error.

@thinkyhead - Do you have any idea why this would only manifest *after* the printer has been powercycled?

CRCinAU, 2021-06-19:
Ok, decided not to sleep tonight to track this down..... Going through the commits on the 9th:

Building from:
https://github.com/MarlinFirmware/Marlin/commit/70b80fcd16722ecbd8ada54c0fb867041cb5f1fd <-- works fine.
https://github.com/MarlinFirmware/Marlin/commit/7c5e3b9071f2ca108ae636dff72b9558479950e4 <-- seems ok (which blows the EEPROM theory out of the water)
https://github.com/MarlinFirmware/Marlin/commit/77496c8235dab5182312652b4dc0b6d0f182cf30 <-- goes boom.

CRCinAU, 2021-06-20:
Testing PR https://github.com/MarlinFirmware/Marlin/pull/22184 by @thinkyhead 

Ran multiple test cycles - could not reproduce the MINTEMP failure with this patch applied.

CRCinAU, 2021-06-21:
Fixed after the merge of https://github.com/MarlinFirmware/Marlin/pull/22184

closed 2021-06-21 by CRCinAU