MarlinFirmware/Marlin #22154: [BUG] How to send an led M150 when error shows

tjthejuggler, 2021-06-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This isn't exactly an issue, I'm sorry, I just don't know where else to ask. We have spent so many hours trying to send a M150 command to change the LED color at line 1298 of temperature.cpp where it says

_temp_error(heater_id, str_t_heating_failed, GET_TEXT(MSG_HEATING_FAILED_LCD));

and we are so stuck. We just want the lights to change when that message pops up. I feel like it shouldn't be that hard, but I also feel like we have tried everything. Can someone please tell me what we need to put there to do this? Thank you so much! Any help is greatly appreciated!

### Bug Timeline

New

### Expected behavior

Send led command

### Actual behavior

Can't send led command

### Steps to Reproduce

No steps

### Version of Marlin Firmware

Most recent

### Printer model

Ender 3

### Electronics

Mks

ellensp, 2021-06-17:
off the top of my head, this should work.

```DIFF
diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
index 9274d0631e..2b91e8d297 100644
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -665,8 +665,11 @@ volatile bool Temperature::raw_temps_ready = false;
                 temp_change_ms = ms + SEC_TO_MS(watch_temp_period);     // - move the expiration timer up
                 if (current_temp > watch_temp_target) heated = true;  // - Flag if target temperature reached
               }
-              else if (ELAPSED(ms, temp_change_ms))                   // Watch timer expired
+              else if (ELAPSED(ms, temp_change_ms)) {                  // Watch timer expired
+                  char buf[10] = "M150 R255";
+                  queue.enqueue_one_now(buf);
                 _temp_error(heater_id, str_t_heating_failed, GET_TEXT(MSG_HEATING_FAILED_LCD));
+              }
             }
             else if (current_temp < target - (MAX_OVERSHOOT_PID_AUTOTUNE)) // Heated, then temperature fell too far?
               _temp_error(heater_id, str_t_thermal_runaway, GET_TEXT(MSG_THERMAL_RUNAWAY));

```

tjthejuggler, 2021-06-17:
> off the top of my head, this should work.
>
> ```diff
> diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
> index 9274d0631e..2b91e8d297 100644
> --- [...]

Thank you SO much for the reply! Unfortunately we tried and it didn't work. I'm thinking(guessing) maybe it also needs a forceCompleteUpdate() after it because some other places this is done that gets used, but it says it is not defined(even though we included the file that holds that function). İs there anything else you can think of? Thanks again, I'm so happy to have gotten a response and to have hope!

tjthejuggler, 2021-06-17:
İt worked!!! Omg! İt totally worked! You are a genius! Thank you so much! The reason it didn't work before is because we were testing the wrong error. Your code is perfect, i can't thank you enough, i don't know what to say, we've spent so many hours on this! Thank you thank you thank you!

tjthejuggler, 2021-06-17:
Ok, so, i have discovered that it sort of works! The only thing is, it only works if i comment out the error message that is sent right there with it. İf i do both together, then only the error message goes, not the led color change. Any idea why this might be? Thanks again, you don't know how helpful this is! İ really appreciate everything you've done with this project, and helping me!

ellensp, 2021-06-18:
Due to it being an error, it can stop before the gcode code is processed.

so try this, it sets the leds directly, no queuing  

```DIFF
+++ b/Marlin/src/module/temperature.cpp
@@ -665,8 +665,10 @@ volatile bool Temperature::raw_temps_ready = false;
                 temp_change_ms = ms + SEC_TO_MS(watch_temp_period);     // - move the expiration timer up
                 if (current_temp > watch_temp_target) heated = true;  // - Flag if target temperature reached
               }
-              else if (ELAPSED(ms, temp_change_ms))                   // Watch timer expired
+              else if (ELAPSED(ms, temp_change_ms)) {                  // Watch timer expired
+                leds.set_color(255,0,0);
                 _temp_error(heater_id, str_t_heating_failed, GET_TEXT(MSG_HEATING_FAILED_LCD));
+              }
             }
             else if (current_temp < target - (MAX_OVERSHOOT_PID_AUTOTUNE)) // Heated, then temperature fell too far?
               _temp_error(heater_id, str_t_thermal_runaway, GET_TEXT(MSG_THERMAL_RUNAWAY));
```

tjthejuggler, 2021-06-18:
ellensp! Thank you so much, this works perfectly. İ was sure i had tried this, i don't know what i did different before. We really appreciate your help with this!

closed 2021-06-18 by tjthejuggler