MarlinFirmware/Marlin #22155: [FR] (Noconstant extrudion - variable feedrate)

mrmun3, 2021-06-17:
### Is your feature request related to a problem? Please describe.

Sorry to ask like this but i couldnt find anything on this topic.
Imagine this. You have layer hight H and you want to make slope to fill between layer n-1 and layer n after layer n is printed. You would have to make triangle then rectangle (rectangle becouse of nozzle width - ½ of width - to avoid collision - see picture) As fas as i understand, comand G1 for extuder just puts constat flow rate to extruder. For triangle part i would need to have variable flow starting with "0" and finishing with flow that would gave me volume od layer height times line width times nozzle width. I know this math is problem for slicer, but in order to make this happen we need to have lineary increasing or decreasing flow of material.

[image]

### Describe the feature you want

Lineary variable flow 
Instead flow = const to have flow=const*path or distance, or coodinate etc.

XDA-Bam, 2021-06-20:
This is an interesting idea. Maybe it will become helpful for non-planar slicing in the future.

Right now, there is a simpler alternative: Approximate the slope of your triangle by using multiple shorter blocks with different height/extrusion rate. Thereby, instead of a smooth slope, you will get a number of very small steps approximating hat slope. If you use enough subdivisions (which can have disadvantages), you can even outresolve the step size of your motors, which will effectively deliver the same printed result as your suggestion.

This is the same discretization approach used for everything else related to movements in Marlin and should be good enough for almost all practical applications. It should be implemented in the slicer.

mrmun3, 2021-06-20:
> This is an interesting idea. Maybe it will become helpful for non-planar slicing in the future.
>
> Right now, there is a simpler alternative: Approximate the slope of your triangle by using multiple [...]

This si step between non planar and planar printing. I am not that much in controling motors but if you can control acceleration of extruder gear, give it constant rate of acceleration and you will have evrything that you need for making slope. 
If you have slope in lenght L with height difference H you need to have constant z speed and constant planar (xy) speed, than you can give extruder feed rate by calculation. Your slope angle is atan(H/L). Than your extruder speed is lineary progressing with z height extruder feed rate = x * (current Z - starting Z) where x is constant feed rate for layer height H with width W. This problem is a bit more complex if you add acceleration to z movement and acceleration to xy movements. If you wnat i can try to make some calculations but i cant implement them to firmware without knowledge.

thinkyhead, 2021-07-30:
I don't see us adding interpolation to these factors anytime soon. It would be a very deep and fundamental change to the Planner and Stepper classes, and it is not clear that it would be any more valuable than using the variable layer heights and ironing features of modern slicers.