MarlinFirmware/Marlin #22156: [FR] Add or include description for artillery Sidewinder X1 TFT

Sonicboom247, 2021-06-17:
### Is your feature request related to a problem? Please describe.

I want to ensure i am utilizing the capabilities of the TFT to its fullest, currently selecting the REPRAP Discount full Graphics controller works, but the alternating progress does not and the speaker tone adjustment is hit or miss, works on one X1 and not my other.

### Are you looking for hardware support?

Artillery Sidewinder X1 V4
Board is MKS GEN L
TFT i would love to see listed in LCD/TFT is https://www.aliexpress.com/i/4001362391075.html

### Describe the feature you want

https://www.aliexpress.com/i/4001362391075.html added to TFT group for the SideWinder X1 or description to which other TFT can be defined in its place.

thisiskeithb, 2021-06-17:
Those are just like every other serial-connected computer/host, so you only need to ensure you have the proper `SERIAL_PORT*` settings to have the TFT and motherboard talk to each other.

Here's an example [Sidewinder X1](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Artillery/Sidewinder%20X1) config in the Configuration repo.

closed 2021-06-17 by thisiskeithb

Sonicboom247, 2021-06-17:
Thank you @thisiskeithb I do have the correct serial port selected, and if i select REPRAP DISCOUNT FULL LCD GRAPHICS CONTROLLER, which I know it isn't, i compile just fine, however if i follow the Sidewinder X1 example which has no TFT defined, i get these errors.
[image]

Sonicboom247, 2021-06-17:
Thank you @thisiskeithb it's not that i can't get the TFT to function, it is that i have to choose the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, but if i follow the example configs you linked my compile throws these errors. 
[image]

thisiskeithb, 2021-06-17:
> it is that i have to choose the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

The X1 doesn't use 12864 LCD, so I'm not sure why you are enabling it. The linked config was tested by several X1 users without needing to change or enable other LCDs.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Sonicboom247, 2021-06-17:
Thanks again, i will check the fb group, perhaps the config hasn't been tested recently with all the new changes. just recompiled with no tft selected and get the same errors

thisiskeithb, 2021-06-17:
> Thanks again, i will check the fb group, perhaps the config hasn't been tested recently with all the new changes. just recompiled with no tft selected and get the same errors

<details><summary>Something is wrong with your VSCode install / Marlin download because it builds fine:</summary>

```prolog
> Executing task: platformio run <

Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio/build/mega2560/src/src/HAL/AVR/HAL.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/HAL_SPI.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/MarlinSerial.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/Servo.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/eeprom.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/fast_pwm.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/fastio.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/watchdog.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/Delay.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/HAL_MinSerial.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/eeprom_api.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/esp_wifi.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/servo.cpp.o
Compiling .pio/build/mega2560/src/src/MarlinCore.cpp.o
Compiling .pio/build/mega2560/src/src/core/serial.cpp.o
Compiling .pio/build/mega2560/src/src/core/utility.cpp.o
Compiling .pio/build/mega2560/src/src/feature/ammeter.cpp.o
Compiling .pio/build/mega2560/src/src/feature/babystep.cpp.o
Compiling .pio/build/mega2560/src/src/feature/leds/leds.cpp.o
Compiling .pio/build/mega2560/src/src/feature/leds/printer_event_leds.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/calibrate/G28.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M200-M205.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M220.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M221.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M301.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M302.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M304.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M309.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M92.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M108_M112_M410.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M111.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M120_M121.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M17_M18_M84.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M211.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M400.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M80_M81.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M85.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M993_M994.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M997.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M999.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/T.cpp.o
[... 80 lines cut ...]
Compiling .pio/build/mega2560/FrameworkArduino/wiring_analog.c.o
Compiling .pio/build/mega2560/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/mega2560/FrameworkArduino/wiring_pulse.S.o
Compiling .pio/build/mega2560/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/mega2560/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/mega2560/libFrameworkArduino.a
Indexing .pio/build/mega2560/libFrameworkArduino.a
Linking .pio/build/mega2560/firmware.elf
Checking size .pio/build/mega2560/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  35.7% (used 2923 bytes from 8192 bytes)
Flash: [===       ]  26.6% (used 67518 bytes from 253952 bytes)
Building .pio/build/mega2560/firmware.hex
================================ [SUCCESS] Took 11.49 seconds ================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:00:11.485
================================ 1 succeeded in 00:00:11.485 ================================

Terminal will be reused by tasks, press any key to close it.
```
</details>

Again, please use the links in my other comment for further troubleshooting.

Sonicboom247, 2021-06-18:
thank you again, i was able to figure it out from one of my errors that mentioned something about SD_SUPPORT, i commented it out and walla, build success, funny enough i learned with SD_SUPPORT and REPRAP full discount controller defined, everything on the printer worked just fine.  will test tomorrow if progress bad alternates with current settings.  THANKS AGAIN