MarlinFirmware/Marlin #22157: [BUG] Ender 3 Pro, SKR 1.3 slow boot time

valemaio2, 2021-06-17:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

On the latest stable release, the boot up time for the printer is of around 1 minute, where as with version 2.0.7.2 it was of around 10 seconds or so.

### Steps to Reproduce

1. Turn on the printer
2. Wait for the printer to boot
3. Go put the kettle on as it's taking too long

### Version of Marlin Firmware

2.0.9

### Printer model

Ender 3 Pro

### Electronics

SKR v1.3, TMC2209 drivers

### Add-ons

Pin 2_04 (extruder 1) used to power hotend fan, to enable automatic fan shutdown below 50°C

### Host Software

OctoPrint

### Additional information & file uploads

[Config.h and config_adv.h attached.](https://github.com/MarlinFirmware/Marlin/files/6671773/Marlin.zip). All other files are unchanged from stock Marlin.

ellensp, 2021-06-18:
delete all the stuff off of your sdcard. It scans the sdcard on startup looking for firmware.bin

slowbro, 2021-06-27:
@valemaio2 did you try booting without an SD card inserted?

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2021-11-25:
Closing, based on lack of reply from the submitter despite multiple suggestions to help investigate the issue.

closed 2021-11-25 by sjasonsmith