MarlinFirmware/Marlin #22160: [BUG] STM32F103RET6_creality fails to compile with Ender-3 V2 default config

rfinnie, 2021-06-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

STM32F103RET6_creality fails to compile with with Ender-3 V2 default config. STM32F103RET6_creality_maple compiles successfully, however.

```
ryan@linda:~/git/Marlin{bugfix-2.0.x}$ git reset --hard
HEAD is now at d0df8bc3cb 💡 Add G28 L description (#22144)
ryan@linda:~/git/Marlin{bugfix-2.0.x}$ git -C ~/git/Marlin-Configurations show | head
commit 415943fa3f11384b0b1366dab71129898cf15a1f
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Mon Jun 14 17:39:29 2021 -0500

    ✨ REDUNDANT_PART_COOLING_FAN
    
    MarlinFirmware/Marlin#21888

diff --git a/config/default/Configuration_adv.h b/config/default/Configuration_adv.h
index 49a0b7b13..ed2ccfdc7 100644
ryan@linda:~/git/Marlin{bugfix-2.0.x}$ cp ~/git/Marlin-Configurations/config/examples/Creality/Ender-3\ V2/Configuration* Marlin/
ryan@linda:~/git/Marlin{bugfix-2.0.x}$ platformio run --environment STM32F103RET6_creality --target clean
Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Build environment is clean
================================================================================ [SUCCESS] Took 0.21 seconds ================================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  SUCCESS   00:00:00.207
================================================================================ 1 succeeded in 00:00:00.207 ================================================================================
ryan@linda:~/git/Marlin{bugfix-2.0.x}$ platformio run --environment STM32F103RET6_creality
Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerialM> 1.0.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/PeripheralPins.c.o
[snip]
Compiling .pio/build/STM32F103RET6_creality/src/src/libs/buzzer.cpp.o
Marlin/src/lcd/dwin/e3v2/dwin.cpp:497:71: warning: use of 'auto' in parameter declaration only available with '-fconcepts'
  497 | inline bool Apply_Encoder(const ENCODER_DiffState &encoder_diffState, auto &valref) {
      |                                                                       ^~~~
Compiling .pio/build/STM32F103RET6_creality/src/src/libs/crc16.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/libs/numtostr.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/libs/stopwatch.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/libs/vector_3.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/module/endstops.cpp.o
Compiling .pio/build/STM32F103RET6_creality/src/src/module/motion.cpp.o
[... 84 lines cut ...]
1 ryan@linda:~/git/Marlin{bugfix-2.0.x}$ 

```

### Bug Timeline

As of 2.0.9, presumably before, during _maple split

### Expected behavior

Successful compilation :)

### Actual behavior

```
*** [.pio/build/STM32F103RET6_creality/liba2a/SoftwareSerialM/HAL_softserial_STM32.cpp.o] Error 1
```

### Version of Marlin Firmware

bugfix-2.0.x HEAD d0df8bc3cb

### Printer model

Ender 3 V2

The-EG, 2021-06-17:
Delete your entire `.pio` directory in your Marlin source directory and try building again.

rfinnie, 2021-06-17:
... huh, that worked, despite not even having a STM32F103RET6_creality build directory, so I can't see why there was a problem before.  I'll close, but I'm confused. Thanks!

closed 2021-06-17 by rfinnie

The-EG, 2021-06-17:
Many build environments, including this one, were changed from maple -> stm32 without being renamed (the 'old' environment is now *_maple). Even if you 'build clean', the dependencies for the 'old' environment were cached and since the 'new' environment had the same name, it was trying to build them.