MarlinFirmware/Marlin #22161: Missing ENC_BTN in pins file when i use SPI COLOR TFT with BTT GTR

djdoubleg, 2021-06-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Utolsó sor nélkül nem lehet használni a gombot.
#if ANY(TFT_COLOR_UI, TFT_LVGL_UI, TFT_CLASSIC_UI)

  #define TFT_CS_PIN                 EXP2_07_PIN
  #define TFT_A0_PIN                 EXP2_04_PIN
  #define TFT_SCK_PIN                EXP2_09_PIN
  #define TFT_MISO_PIN               EXP2_10_PIN
  #define TFT_MOSI_PIN               EXP2_05_PIN

  #define TOUCH_INT_PIN              EXP1_04_PIN
  #define TOUCH_MISO_PIN             EXP1_05_PIN
  #define TOUCH_MOSI_PIN             EXP1_08_PIN
  #define TOUCH_SCK_PIN              EXP1_06_PIN
  #define TOUCH_CS_PIN               EXP1_07_PIN
  #define BTN_EN1                    EXP2_08_PIN
  #define BTN_EN2                    EXP2_06_PIN
  #define BTN_ENC                    EXP1_09_PIN <------ this line is missing in 2.0.9 GTR pins config

### Bug Timeline

NEW (new pin config file)

### Expected behavior

I expect the push button would work.

### Actual behavior

The knob button did not work

### Steps to Reproduce

Compile the code
Update the FW.
Then the knob not work.

### Version of Marlin Firmware

2.0.9

### Printer model

Self designed with BTT GTR

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

Insert this line to TFT defined section.
  #define BTN_ENC                    EXP1_09_PIN

thisiskeithb, 2021-06-17:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

looxonline, 2021-06-17:
Grubits, don't you learn not to mention unreleased products?

djdoubleg, 2021-06-17:
> 
> 
> Grubits, don't you learn not to mention unreleased products? If you have a fix, thats great. Use it, but there is no need to mention a product that has still not been announced by the manufacturer in your report.

What do you think?
Show me where I mention anything?

djdoubleg, 2021-06-17:
i use this SPI COLOR TFT with external rotry switch. 
[image]
Please explain the accusation.

thisiskeithb, 2021-06-17:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22162

closed 2021-06-17 by thisiskeithb

looxonline, 2021-06-18:
> > Grubits, don't you learn not to mention unreleased products? If you have a fix, thats great. Use it, but there is no need to mention a product that has still not been announced by the manufacturer in your report.
> 
> What do you think?
> Show me where I mention anything?

You are correct. I misread the title of the report and typed too fast before going to bed. My apologies.

PR #22162 (2021-06-18): TFT encoder pin for BTT GTR
### Description

Define TFT encoder pin for BTT GTR.

### Benefits

BTT GTR with a compatible TFT that runs `TFT_COLOR_UI`, `TFT_LVGL_UI`, or `TFT_CLASSIC_UI` can use an encoder.

### Related Issues

Fixes #22161