MarlinFirmware/Marlin #22166: [BUG] 2.0.9 : E1 stepper not working when switching tool T0 -> T1

snakesrules94, 2021-06-18:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

E1 stepper not working when switching tool T0 -> T1

### Bug Timeline

new bug

### Actual behavior

E1 stepper should turn

### Version of Marlin Firmware

2.0.9

### Printer model

Custom

### Electronics

Robin nano V2

### Additional information & file uploads

Hi there, switching from release 2.0.8.2 to 2.0.9 and i have a problem when switching tool . i have a single nose with 2 entries and two extruders .
In 2.0.8.2 , if i switch tool T0 to T1 or apply a mix , everything is ok .
I've report the same config to the latest 2.0.9 release and E1 stepper never turns , in switching tool mode or in mix mode .

I will roll back to 2.0.8.2 until further correction.

Regards.

ellensp, 2021-06-18:
Please attach your 2.0.9 configuration files so we can test.

snakesrules94, 2021-06-18:
Hello , here are the config files.
Regards.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6678149/Marlin.zip)

slowbro, 2021-06-22:
> do you have link to 2.0.8.2

Here it is: https://github.com/MarlinFirmware/Marlin/tree/2.0.8.2

nayim12, 2021-06-22:
I have same pblm

SkyTakes, 2021-06-27:
Exactly same problem! Worked perfectly up to 2.0.8.2
Board: BOARD_ZRIB_V20
Mixing Extruder - E0 OK / E1 Not working
If I change the config to:
`#define EXTRUDERS 2` (set 2)
`#define SINGLENOZZLE` (uncomment)
`// #define MIXING_EXTRUDER` (comment)
then E0 and E1 works but I´m not able to mix (as it should be)

PS.  Just tried 2.0.9.1 - same problem - E1 not working if mixing extruder is selected

Robo-Print, 2021-06-29:
I have the same problem with a Geeetech A10M using Marlin  2.0.9, 2.0.9.1 or bugfix-2.0.x version that only the first extruder is working at Mixing-Extruder mode.
I tried it with A4988 and also TMC2209 V3.1 stepper drivers.

Switching from Mixing-Extruder mode to Single-Nozzle mode with two extruders both extruders work as SkyTakes already mentioned.
Also if I am using an older Marlin 2 bugfix version from november 2020 (I used before upgrading to 2.0.9) both extruder work in Mixing-Mode without any problem.

slowbro, 2021-07-07:
@Robo-Print when in mixing mode, is the second extruder online, i.e. held in place by the driver, or can you freewheel it?

Robo-Print, 2021-07-07:
Hello slowbro,
In mixing mode the second extruder is held in place, I can‘t turn the wheel of the extruder.

thinkyhead, 2021-07-30:
Mixing has not been intentionally touched for quite some time. Please try to narrow down the exact commit where the problem first appears.

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

snakesrules94, 2021-07-30:
@thinkyhead : hello , i've just checked and tried the bugfix release , behavior is still the same : E1 motor does not turn after selecting T1 and asking for extrusion . The motor is not locked during the situation , i can manually turn it .

Robo-Print, 2021-07-30:
The same at my Geeetech A10M printer. I used the actual version of the bugfix-2.0.x branch and the second extruder in mixing mode is also not working. But I am also able to move the second extruder, it is not locked.

Sorry in contrast to my comment of july 7th I also can turn the second Extruder during and after print attempts in mixing mode, I only need a little bit more force than I expected and now I saw that I can't turn the wheel of the first extruder at all but the second with more force...

thinkyhead, 2021-07-31:
It sounds like ENABLE signals are getting lost.

Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until the exact day where it broke is found.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

snakesrules94, 2021-07-31:
> 
> Would you be able to test some `bugfix-2.0.x` commits to narrow this down?
> 
Hello,
sure , will do it starting from 1st of june , just after the 2.0.8.2 release to now .

snakesrules94, 2021-07-31:
first check this afternoon : 

- commit 74be64a from 01/06 ==> ok , both steppers are turning
- commit 03a4697 from 30/06 ==> NOT OK , only stepper 1 is turning

To be continued ...

snakesrules94, 2021-08-01:
Hello,
seems i've found something : 

`commit 6131d2c5a6285da849fbb8d3e592f89518b02b24
Author: ldursw <37294448+ldursw@users.noreply.github.com>
Date:   Fri Jun 4 00:38:10 2021 -0300

    🔨 MKS Robin E3 for HAL/STM32 (#21927)`

result OK , both steppers are turning.

Next commit in log : 

`commit eea9b6d8ae38af3ca9d5b2f6fb355fc8fce405fb
Author: Giuliano Zaro <3684609+GMagician@users.noreply.github.com>
Date:   Sat Jun 5 03:02:37 2021 +0200

    🐛 Fix MMU compile with >5 EXTRUDERS (#22036)`

result NOT OK , only steppers 1 (T0) is turning .

snakesrules94, 2021-08-01:
Hi again , i've made a quick & dirty test : 
took the latest bugfix , reverted the changes in planner.h and planner.cpp from commit eea9b6d and both steppers behavior seems ok ( quick test with an m302 P1 followed by a 5mm extrude with T0 then T1 just to see the steeper turning ) .

Markeydoo, 2021-08-02:
Having the same issue in 2.9.1.
If I set #define EXTRUDERS 2 and #define SINGLENOZZLE both T0 and T1 work but will not Mix.
If I set #define EXTRUDERS 1 and //#define SINGLENOZZLE and #define MIXING_EXTRUDER, T0 works but T1 never engages.
Thanks

snakesrules94, 2021-08-02:
Hello, i guess there is something related in the DISABLE_INACTIVE_EXTRUDER modification ,not re-engaging the second stepper when selected.

Markeydoo, 2021-08-02:
I have it working with 2.0.8.2 thanks
Does anyone have it working with 2.0.9.1?

snakesrules94, 2021-08-03:
> 
> 
> I have it working with 2.0.8.2 thanks
> Does anyone have it working with 2.0.9.1?

Hello , you could always try this : https://github.com/MarlinFirmware/Marlin/issues/22166#issuecomment-890488765
but as i' don't know the potential side effects , i would not recommend it .

Vertabreak, 2021-08-07:
same issue with the A20T mixing_extruder, E0 works but as soon as you change vtools to anything none of them work.

thinkyhead, 2021-08-07:
Yeah, I have been looking with suspicion at `DISABLE_INACTIVE_EXTRUDER` for some time. I'll also look at the changes from the referenced commit above.

Markeydoo, 2021-08-13:
Unable to Disable_Inactive_extrder with Mixing hot end.

It is working in 2.0.8.2. Mixing Hotend, E0 and 1 both work, I have also installed BLTouch and 2 Smart Filament Sensors and all are working.
I am also able to control E0 and E1 and Bltouch through my TFT35.
Mark

Vertabreak, 2021-08-13:
//DISABLE_INACTIVE_EXTRUDER 
corrected the issue on all my machines with mixing_extruder.

Markeydoo, 2021-08-13:
What was changed from version 2.0.8.2 to 2.0.9.1?

rrroonn, 2021-08-14:
Affects me too.  SWITCHING NOZZLE.  Seems like the same issue to me.

If (via LCD) I select the second extruder and then do an extrusion (E only move), the extruder does move a short distance but then stops and will not move further.  If I then (via LCD) try to jog the X axis, it also does a short move and then stops.  After about 30 seconds (timeout?), the X axis then seems to free itself up and completes the X move.

Seems like major changes to handling (additional axis) and something in E got missed.

DerAndere1, 2021-12-16:
To track down the error further: does it help to change the following in current bugfix-2.0.x to get mixing back?
https://github.com/MarlinFirmware/Marlin/blob/6dc056f77114a0c3e2cf68758eaa7cf41061e173/Marlin/src/module/planner.cpp#L2243

change it to
```
 REPEAT(EXTRUDERS, ENABLE_ONE_E); // (ENABLE_ONE_E must end with semicolon) 
```
In addition, do not define `DISABLE_INACTIVE_EXTRUDER`, so you should have in Configuration_adv.h:
```
//#define DISABLE_INACTIVE_EXTRUDER
```

closed 2022-02-26 by github-actions[bot]