MarlinFirmware/Marlin #22167: [BUG] (bug summary)

HoverClub, 2021-06-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Multiple Compiler errors similar to: 
Marlin/src/module/endstops.cpp: In static member function 'static void Endstops::update()':
Marlin/src/module/endstops.cpp:623:25: error: 'X_HEAD' was not declared in this scope
     #define X_AXIS_HEAD X_HEAD

### Bug Timeline

Today.

### Steps to Reproduce

1. Set Machine type to MarkForged_XY
2. Set axis to 3
3. Complie

### Version of Marlin Firmware

2.0.9

### Electronics

MKS Gen-L

### Additional information & file uploads

Editing line 86 of types.h seems to fix the issue (not tested on a real machine yet!).
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6678304/config.zip)

Change:
 #if IS_CORE  
 to:
 #if EITHER(IS_CORE, MARKFORGED_XY)

DerAndere1, 2021-06-18:
This regression was introduced in commit https://github.com/MarlinFirmware/Marlin/commit/dd4990252e891cdfe56cb7d6e3bbe1e6289be649 . The proposed fix ssems reasonable. Please report your findings when you have tested it in hardware.

thinkyhead, 2021-06-18:
Patched. Thanks!

commit 55feb8ac (2021-06-18): 🐛 Define 'HEAD' axes for Markforged

Fixes #22167
commit cce585f6 (2021-06-18): 🐛 Define 'HEAD' axes for Markforged

Fixes #22167
closed 2021-06-18 by thinkyhead