MarlinFirmware/Marlin #22168: [BUG] Compile error for numberless, non-animated, non-inverted lcd hotends

codingcatgirl, 2021-06-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When hotends are configured to be numberless, not animated and not inverted, marlin fails compilation.

### Expected behavior

Marlin should compile.

### Actual behavior

This specific combination of settings is not covered in the Compile Macros and Marlin compilation fails with error:
```
In file included from Marlin/src/lcd/dogm/../../inc/MarlinConfigPre.h:37:0,
                 from Marlin/src/lcd/dogm/status_screen_DOGM.cpp:28:
Marlin/src/lcd/dogm/status_screen_DOGM.cpp: In function 'void _draw_hotend_status(heater_id_t, bool)':
Marlin/src/lcd/dogm/status_screen_DOGM.cpp:249:26: error: 'status_hotend1_a_bmp' was not declared in this scope
       #define OFF_BMP(N) status_hotend##N##_a_bmp
                          ^
```

### Steps to Reproduce

1. Enable LCD
2. Enable `STATUS_HOTEND_NUMBERLESS`
3. Disable `STATUS_HOTEND_INVERTED`
4. Disable `STATUS_HOTEND_ANIM`
5. Try to compile, marlin tries to get the numbered hotend images but they don't exist

### Version of Marlin Firmware

2.0.9

codingcatgirl, 2021-06-18:
Update: A similar, but different error is thrown when STATUS_HOTEND_INVERTED is enabled.

ellensp, 2021-06-20:
@codingcatgirl could you test the Pr

codingcatgirl, 2021-06-20:
I shall once i get the chance, possibly later today :) Thanks for your work!

ellensp, 2021-06-22:
This has now been merged

closed 2021-06-22 by ellensp