MarlinFirmware/Marlin #22169: [BUG] Complie error for MAX_FEEDRATE_EDIT_VALUES or MAX_ACCEL_EDIT_VALUES

HoverClub, 2021-06-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Enabling LIMITED_MAX_FR_EDITING and setting MAX_FEEDRATE_EDIT_VALUES  (same for LIMITED_MAX_ACCEL_EDITING) throws compile errors when values set to same number of entries as steppers.

`../inc/SanityCheck.h:3200:5: error: static assertion failed: MAX_FEEDRATE_EDIT_VALUES has too many elements. X Y Z E elements only.
`

Looks like it's probably comparing to the wrong array length in SanityCheck but I'm not sure if that's the cause - being unfamiliar with the code!
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6678970/config.zip)

### Bug Timeline

Today

### Steps to Reproduce

1. In Configuration.h uncomment #define LIMITED_MAX_FR_EDITING
2. Set MAX_FEEDRATE_EDIT_VALUES for each axis.
3. Complie 

### Version of Marlin Firmware

2.09

ellensp, 2021-06-19:
Never mind, I missed them, they are there.

ellensp, 2021-06-19:
@DerAndere1 could you take a look at this?

slowbro, 2021-06-19:
This also happens if `MAX_FEEDRATE_EDIT_VALUES` is not defined (just `DEFAULT_MAX_FEEDRATE`) - looking at the code, I'm not sure if the functions that use those were ever modified for `DISTINCT_E_FACTORS`.. for example,

https://github.com/MarlinFirmware/Marlin/blob/758d495d08327dd28e8e6b0880f77da0b140def2/Marlin/src/lcd/menu/menu_advanced.cpp#L353-L361

It's initializing a `xyze_feedrate_t` which doesn't expect to get multiple E values. That seems to be used for scaling the values in the case that `MAX_FEEDRATE_EDIT_VALUES` is not defined.

DerAndere1, 2021-06-19:
Yes that is why I did not just do a quick PR to fix compilation. i guess this was introduced with commit https://github.com/MarlinFirmware/Marlin/commit/dd4990252e891cdfe56cb7d6e3bbe1e6289be649 so we have to review those changes and see how `DISTINCT_E_FACTORS` should be handled. 
While we are at it, SENSORLESS_HOMING with EXTRUDERS 5 should also be fixed. (Related issue: https://github.com/MarlinFirmware/Marlin/issues/22012)

DerAndere1, 2021-06-22:
This is not a bug. You can change this issue to a feature request if you want to set distinct values for individual E steppers of MAX_FEEDRATE_EDIT_VALUES or MAX_ACCEL_EDIT_VALUES  when DISTINCT_E_FACTORS is enabled.

@HoverClub: You have to obey the compiler error messages from SanityCheck.h: remove the last element from the arrays MAX_FEEDRATE_EDIT_VALUES and MAX_ACCEL_EDIT_VALUES.

HoverClub, 2021-06-23:
OK thanks.  Maybe it should be called an "inconsistency" in that you can set default feed rate and accel values for all axes but can only edit some of them?

closed 2021-06-23 by HoverClub