MarlinFirmware/Marlin #22170: [BUG] (bug summary)

qulet, 2021-06-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Płyta MKS Sbase  V.1.4 REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER  SD not work.
Custom cabel:
[README.txt](https://github.com/MarlinFirmware/Marlin/files/6679577/README.txt)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6679580/Marlin.zip)

### Bug Timeline

Mks Sbase, New core XY compilation

### Expected behavior

I expected that the special cable described in "readme" would allow reading the files stored on the SD card

### Actual behavior

Entering the card reading shows the message refresh, clicking does not change anything ...

### Version of Marlin Firmware

02000900

### Printer model

DIY Core XY

### Electronics

MKS SBASE v1.4

### Add-ons

special cable

### Your Slicer

Simplify3D

slowbro, 2021-06-20:
@qulet Please replace "(bug summary)" in the title with a more descriptive title.

qulet, 2021-06-20:
used to go to P0.8 to J8-3
     "" "" P0.7 is J8-2
     "" "" P0.5 is J8-4
     P0.27 is "" "" J8-5
In the Readme file there is a special cable error. Pin P0.5 is the connector of the X axis stepper motor.

My special cable made:
                        P0.8 is J8-3
     "" "" P0.7 is J8-2
     "" "" P0.9 to J8-4 ***
     P0.27 is "" "" J8-5
**** The SD card on my special cable doesn't work either ......
compilation done without file changes:
pins_MKS_SB.h
spi_pins.h
Maybe someone will finally solve this problem with reading the SD card built into the FGSC ....... and publish how to do it properly.

qulet, 2021-06-21:
Marlin taken from the source MKS-SBASE-master.zip
cannot be compiled

qulet, 2021-06-22:
Recipe for a working SD LCD FGSC card reader MKS-SBASE board:

Custom cable
Inverted 2x tape plugs (180 degrees)
       EXP2 S-Base FGSC EXP2
         P0.8 to is J8-3 SD MISO
         P0.7 to is J8-2 SD SCK
         P0.9 to is J8-4 SD MOSI
        P0.27 to is J8-5 SD DEDECT

Need to change the Configuration_adv.h file:
**
    * Set this option to one of the following (or the board's defaults apply):
    *
    * LCD - Use the SD drive in the external LCD controller.
    * ONBOARD - Use the SD drive on the control board.
    * CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
    *
    *: ['LCD', 'ONBOARD', 'CUSTOM_CABLE']
    * /
   #define SDCARD_CONNECTION CUSTOM_CABLE 
   
   Tested on Marlin Marlin-bugfix-2.0.9

closed 2021-06-22 by qulet