MarlinFirmware/Marlin #22175: [BUG] UBL: G29 J mostly appears to nullify bed leveling

salfter, 2021-06-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The print bed in my Hypercube 300 is perhaps a bit more unlevel than is optimal, but I've had UBL to compensate for that in the past.  Lately, though, it appears to not work properly.  During a print of an object of nearly any size, there should be a noticeable amount of movement in the Z axis as X and Y move along...but lately, there isn't.  This has a deleterious effect on first-layer quality of all but trivially small objects.

### Bug Timeline

new...probably noticed it within the past 2-3 weeks

### Expected behavior

Z should move with X and Y to compensate for an unlevel bed.

### Actual behavior

Z hardly moves at all.

### Steps to Reproduce

1. rebuild Marlin with extra debugging options (```DEBUG_LEVELING_FEATURE``` and ```M114_DETAIL```)
2. send the attached gcode files from Pronterface, and capture the communication with the printer (see [test.zip](https://github.com/MarlinFirmware/Marlin/files/6680116/test.zip))

The attached gcode files are "dry runs" that have had heater and extruder commands filtered out.  They will move the printhead in large circles within the print volume of my Hypercube 300; they probably shouldn't be sent to a printer with a smaller than 300x300mm print bed.

Config files: [marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/6680121/marlin-config.zip)

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jun 18 2021 22:04:38) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin

### Printer model

Hypercube 300

### Electronics

SKR 1.4 Turbo with TMC2130 steppers on X/Y/Z/E

### Add-ons

Tiny-Touch NMPG sensor (https://www.thingiverse.com/thing:4661556), though a BLTouch was behaving similarly

### Your Slicer

Prusa Slicer

### Host Software

Other (explain below)

### Additional information & file uploads

Printing host appears to make no difference.  Whether printing from an SD card in the motherboard, an SD card in the TFT35 display (which behaves like a serial-connected host), or printing over a Bluetooth connection from OctoPrint or Pronterface makes no difference.

Looking at the logfiles, I noticed that the messages from get_z_correction() indicate values over a fairly wide range in the printout where G29 J was not used, but that in the one where it was used, the returned values are all gathered around my printer's probe Z offset setting (-0.28 mm).  I noticed when using OctoPrint's Bed Level Visualizer, G29 L1 would show the expected tilt in the bed, but refreshing the visualizer after G29 J would show a flat plane below the zero point.

DerAndere1, 2021-06-19:
So to sum up some issues related to G29 that cropped up around the time when Marlin was prepared for 6 axis support:
- This issue:  `G29 J` with `AUTO_BED_LEVELING_UBL` for 3point leveling does not work but no mention of non-moving axes X and Y after the G29.
- Possibly related: https://github.com/MarlinFirmware/Marlin/issues/21887
- issue https://github.com/MarlinFirmware/Marlin/issues/22128: AUTO_BED_LEVELING_3POINT: G29 makes movement of axes with G1 impossible.
:-(

DerAndere1, 2021-06-19:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

    Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
    Connect to your printer from host software such as Cura, Printrun or Repetier Host.
    Send M502 and M500 to ensure your Configurations are applied.
    Issue the command M111 S247 to enable maximum logging.
    Perform a G28 to do your standard homing procedure.
    Do a G29 to probe the bed. This will also enable bed leveling.
    Do some of the moves that revealed problems before. Take notes.
    Copy the log output into a .TXT file and attach it to your next reply.

When you have more time, would you like to try to test Marlin before the addition of 6 axis support and report if it works better? That way we could find the cause for this bug more easily. Here is a link to the source code of that Marlin version: https://github.com/MarlinFirmware/Marlin/tree/733d5fd57d6343fcd8643d322129845b57945cad .

salfter, 2021-06-19:
> we now ask everyone to follow a standard procedure

I've done all that already...results are in test.zip, attached above.

> When you have more time, would you like to try to test Marlin before the addition of 6 axis support and report if it works better?

I could give that a shot...not sure how much time I'll have today, but I'll keep it in mind.

salfter, 2021-06-20:
>> When you have more time, would you like to try to test Marlin before the addition of 6 axis support and report if it works better?
>
> I could give that a shot...not sure how much time I'll have today, but I'll keep it in mind.

It's running on the older firmware now, and G29 J behaves as expected: [733f5fd.zip](https://github.com/MarlinFirmware/Marlin/files/6681824/733f5fd.zip)

Comango104, 2021-07-09:
Issues started with release 2.0.9.0, rolled back to 2.0.8.2 to get it working again.

MoellerDi, 2021-07-11:
Same here; as kind of workaround I just removed "G29 J" from cura start g-code for now

hamster65, 2021-07-11:
I can confirm that with today's bugfix firmware, I see absolutely no Z movement during print with G29 J.
Since I am having constant issues with Z over/undercompensation I did not notice it before, so I am not sure when it started.
I'll do the next print without G29 J and report back, but I am not sure when that will be. Oddly, except for the back, my print looks far better then with Z moving.

ivan5rov, 2021-07-13:
I can too confirm the same behavior. Tested both in latest bugfix as well as 2.0.9.1. G29 J essentially flattens the mesh by populating the same value everywhere.

DasSmeagle, 2021-07-16:
Yes, G29 J "destroys" the mesh instead of tilting it and it disables leveling.

bwspath, 2021-07-19:
https://github.com/MarlinFirmware/Marlin/pull/22398 that should fix the issue .. was having the same issue

danielcaceresm, 2021-07-19:
Thanks @bwspath I just tested the #22398 change and the G29 J command is working!!!

bwspath, 2021-07-20:
https://github.com/MarlinFirmware/Marlin/commit/154decfc66c30808cfe7320e542fcf90427d7176 has been merged and fixed! time to close the issue I guess

ellensp, 2021-07-20:
"By your command" closing.

closed 2021-07-20 by ellensp

salfter, 2021-08-23:
Got around to updating firmware a little bit ago after having been gone a while...looks like UBL is working properly again.  w00t!