MarlinFirmware/Marlin #22177: [BUG] Error: Unknown environment names 'STM32F103RC_btt_512K' ---- SKR Mini E3 V2 Board compile error

tijoer, 2021-06-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I cannot compile with plattform.io, as I get this error. It seems to be that the variable was removed. I am now unsure which one to use. I could use the following for my SKR Mini E3 V 2.0 board. All of them sound fine to me. Bigtreetech uses STM32F103RC_btt_maple in their example configuration, but I heard that this is an error. Now I am completly confused.

- STM32F103RC_btt_maple
- STM32F103RC_btt_USB_maple
- STM32F103RC_btt_maple
- STM32F103RC_btt_USB
- STM32F103RC_btt
- BIGTREE_BTT002
- BIGTREE_SKR_2

I think the var should not have been removed, but marked as deprecated instead with a documentation how to migrate.

### Bug Timeline

new

### Expected behavior

Either one:

The compiling works fine or the variable had been marked as deprecated and the compiler would link the documentation to the migration information.

### Actual behavior

Error: Unknown environment names 'STM32F103RC_btt_512K'. Valid names are 'mks_robin_e3p, STM32F103RE_btt_maple, STM32F103ZE, esp32, STM32F103RE_btt_USB, STM32F103RC_fysetc, NUCLEO_F767ZI, STM32F103RC_btt_USB_maple, DUE_archim, FYSETC_S6, sanguino1284p_optimized, STM32F070CB_malyan, STM32F103RC_btt_maple, at90usb1286_cdc, melzi_optiboot, malyan_M300, STM32F103RE_btt, rambo, mks_robin_lite, FYSETC_F6, STM32F407VE_black, teensy36, rumba32, BIGTREE_SKR_2, STM32F103VE_longer, BIGTREE_OCTOPUS_V1_USB, mks_robin, STM32F103RE_btt_USB_maple, ARMED, REMRAM_V1, mks_robin_e3, include_tree, mks_robin_nano_v3_usb_flash_drive, BIGTREE_SKR_PRO_usb_flash_drive, STM32F103VE, mega2560ext, STM32F103RC_btt_USB, melzi_optimized, STM32F103RE_maple, BIGTREE_SKR_PRO, mks_robin_pro, LERDGES_usb_flash_drive, trigorilla_pro, BIGTREE_GTR_V1_0, STM32F103RC_meeb, at90usb1286_dfu, LERDGEX_usb_flash_drive, mingda_mpx_arm_mini, mks_robin_mini, jgaurora_a5s_a1, BIGTREE_E3_RRF, mks_robin_e3_maple, STM32F103CB_malyan_maple, LERDGEK_usb_flash_drive, chitu_f103, STM32F103RC_btt, LERDGES, FYSETC_CHEETAH_V20, STM32F070RB_malyan, mks_robin_pro2, melzi, teensy41, BIGTREE_GTR_V1_0_usb_flash_drive, DUE, sanguino644p, melzi_optiboot_optimized, STM32F103RET6_creality_maple, FLYF407ZG, mks_robin_nano35, LPC1768, MightyBoard2560, chitu_v5_gpio_init, LPC1769, mega1280, flsun_hispeedv1, STM32F103RE, Anet_ET4_OpenBLT, linux_native, mks_robin_nano_v3_usb_flash_drive_msc, MightyBoard1280, BTT_SKR_SE_BX, mega2560, mks_robin_lite3, STM32F103CB_malyan, mks_robin_nano_v3, sanguino1284p, STM32F103RET6_creality, teensy35, LERDGEX, DUE_USB, SAMD51_grandcentral_m4, mks_robin_nano35_maple, teensy31, mks_robin_maple, BIGTREE_OCTOPUS_V1, STM32F401VE_STEVAL, STM32F103VE_GTM32, LERDGEK, BIGTREE_BTT002, FYSETC_E4'

### Steps to Reproduce

- Checkout nightly build (19.06.2021).
- Edit platformio.ini with default_envs = STM32F103RC_btt_maple
- Hit compile in plattformio.

### Version of Marlin Firmware

bugfix-2.0.x nightly build 19.06.2021

### Printer model

Ender 3 Pro

### Electronics

SKR Mini E3 V2.0

### Add-ons

BLTouch

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

none

ellensp, 2021-06-19:
STM32F103RC_btt_512K was a horrendous hack, and should never have been allowed.
If you have a STM32F103RC it only has 256K of Flash, using more than that is not recommended, Its not meant to be there, has never been tested, is quit likely a failed  STM32F103RE they they sold as the lesser chip. 
you use STM32F103RC_btt or if you actually have a STM32F103RE chip STM32F103RE_btt

maple is legacy, will be removed soon
_USB is if is you want to mount your sdcard as a usb drive.

closed 2021-06-19 by ellensp

tijoer, 2021-06-19:
Thank you very much for your helpful explanation, as I am new to embedded development and I am still quite lost sometimes.

However would there be a way to add a proper deprecation warning? If you could guide me to an example of other deprecation warnings I could try to work on a pull request.

TracKer, 2021-08-13:
@ellensp I was using it for a long time (and there are more people), and it was stable enough. I bought the card when 512k chip was not yet installed, so now I need to buy a new one?
Are there some specific problems with keeping this one?