MarlinFirmware/Marlin #22180: [BUG] (bug summary) Encoder selecting Ender 3 Stock LCD not Response

agoy507, 2021-06-20:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I was Flash my board using 2.0.9 firmware. But encoder not working.
when i used 2.0.7 firmware, encoder working normal.
I used CR10 stock display LCD.

### Version of Marlin Firmware

2.0.9

### Printer model

Creality ender 3

### Electronics

ESP32 By Felixstorm

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6681564/Marlin.zip)

thisiskeithb, 2021-06-20:
> #define MOTHERBOARD BOARD_ESP32_FS_ENDER3

This is not a valid motherboard. Are you using stock Marlin?

ellensp, 2021-06-20:
Seems to be https://github.com/felixstorm/Marlin/tree/Felix_Ender3_ESP32_2.0.x
Wich is not stock marlin...

agoy507, 2021-06-20:
i used DIY motherboard https://github.com/felixstorm/Creality_Ender_3_ESP32_Board
i using stock marlin 2.0.7 firmware that.s working normal.
but when i upgrade to 2.0.9 with same config pin, encoder not working 

this pin configuration I used
[pins_ESP32_FS_ENDER3.zip](https://github.com/MarlinFirmware/Marlin/files/6681658/pins_ESP32_FS_ENDER3.zip)

ellensp, 2021-06-20:
That pins files does not enable the encoder pins unless REPRAP_DISCOUNT_SMART_CONTROLLER is enabled..
Presuming it uses directly connected io pins for the encoder. this wont work.

agoy507, 2021-06-20:
ok, I get it. thank for advice. i will try to change that,

closed 2021-06-20 by thisiskeithb