MarlinFirmware/Marlin #22181: [BUG] PRUSA_MMU2 prevents endpoint detection, for sensor and sensorles

kursatu, 2021-06-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Build the latest bugfix

- Home X or Y 
- Observe that the movement does not stop and t it hits the end. You need to power down the machine.
- Undefine PRUSA_MMU2 , it works as expected.
- This defect happens with both optical sensor and sensorless homing.

- Additionally obstruct the sensor and type M119, observe that they are triggered.
- Home and it tries to home even though the sensor is already triggered.
- Same thing happens with the sensorless homing. You can't stop the movement by holding the head.
But if you don't define PRUSA_MMU2 both sensor & sensorless homing work fine.

I've initialized the settings all the time while trying to get it to work.
The printer used to work. I tried to go back many commits (up to 3 months) but none of them helped,
I couldn't get to the last working config anymore,
Same issue with sensorless homing only is reported here and in other referenced issues in that defect.
https://github.com/MarlinFirmware/Marlin/issues/22012

### Bug Timeline

It used to work, it seems like a regression, I've tried to go back as far as 3 months but could not get it to work. I couldn't find what/how I made it work.

### Expected behavior

I expected the movement to stop when the endstop is detected.

### Version of Marlin Firmware

SENDING:M115 FIRMWARE_NAME:Marlin bugfix-2.0.x (Jun 19 2021 16:23:47) 

### Printer model

Custom CoreXY

### Electronics

SKR2 

### Your Slicer

Other (explain below)

### Host Software

Pronterface

### Additional information & file uploads

[Marlin_defect.zip](https://github.com/MarlinFirmware/Marlin/files/6681647/Marlin_defect.zip)

thinkyhead, 2021-07-30:
It sounds like some kind of pin or interrupt conflict. Are you using `ENDSTOP_INTERRUPTS_FEATURE` at all? Can you enable `PINS_DEBUGGING` and send `M43` to check for any pin / timer conflicts?

Of course, please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2021-07-30:
I'm not sure how #21953 will play into the issue, as that was a pretty major refactor that resulted in a lot of regressions. I suspect there is some kind of plus-one bug somewhere in that pile, but related to the MMU2. I'll review that PR and see where changes related to MMU2 and/or endstops were made, and see if there are any suspicious-looking `LOOP` or `REPEAT` constructs.

Minims, 2021-07-30:
Hello, I have the same issue and `ENDSTOP_INTERRUPTS_FEATURE`is not enable on my side.
i have an SKR Pro 1.1 on a CoreXY + MMU2S
I have juste tried with the last bugfix2.0 and the result of `M43` is :  

```
M43
         <unused/unknown>                       Input  = 1                       
PIN: PG12        M42 P110         E2_CS_PIN                              Input  = 1
PIN: PG13        M42 P111                                Input  = 1                       
PIN: PG14        M42 P112                                Input  = 1                       
PIN: PG15        M42 P113         E1_CS_PIN                              Input  = 1
PIN: PH0         M42 P3                                  Input  = 0                       
PIN: PH1         M42 P4                                  Input  = 0                       
ok
```

```
M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_probe: open
filament: TRIGGERED
ok
```

thinkyhead, 2021-07-31:
You both have SKR boards in common. I wonder what that could mean.

The MMU2 uses some serial port defined by `MMU2_SERIAL_PORT`, and there are pins associated with that serial port, and maybe something got broken in relation to that. So far my scanning of commits specifically related to the MMU2 are not yielding any clues.

Try defining `MMU2_DEBUG` in your configuration and then there will be extra log output related to the MMU2, and maybe this will give some indication where the MMU2 is interacting with homing.

thinkyhead, 2021-07-31:
As another test, even though your MMU2 won't function with this, try changing your `MMU2_SERIAL_PORT` setting to see if endstops are still broken by the feature. This will at least tell us whether the serial pins are strongly implicated.

JohnnyTheOne, 2021-07-31:
It has to do with 5 extruders not mmu2  https://github.com/MarlinFirmware/Marlin/issues/22012#issuecomment-864384742

Minims, 2021-07-31:
I have already tried to use anoyher serial port and the endstop issue is the same.

But on a prusa style (not corexy) with skr1.4 turbo-tmc2209 sensorless is working with mmu2.

so it might be related to 
- corexy
- 5 extruders

Minims, 2021-07-31:
If i disconnect the mmu2, serial says mmu2 not responding so disabled. 
but sensorless is still not working

thinkyhead, 2021-08-01:
Yeah, someone else hinted that the root of the issue might be more due to `EXTRUDERS 5` than due to the MMU2 code files, and I suppose that is a possibility. It was suggested that the problem started with the 6-linear-axis refactor, which introduced new constructs to deal with multiple axes, multiple extruders, and multiple hotends.

There is a sanity-check that prevents compiling MMU unless `EXTRUDERS` is 5. It might be possible to set `EXTRUDERS` to some other value, then comment-out the sanity-check so it will compile, and then test for endstop problems. If we find that reducing `EXTRUDERS` to 4 is enough to solve the problem, that would reinforce the hypothesis that we have a plus-one error somewhere in that refactor. It's a lot to hunt through, so the more we experiment the easier it will be to pinpoint the problem area of the code.

Minims, 2021-08-08:
> Yeah, someone else hinted that the root of the issue might be more due to `EXTRUDERS 5` than due to the MMU2 code files, and I suppose that is a possibility. It was suggested that the problem [...]

Hello, 

So I have made some tests on the last bugfix this evening.

So my setup is  :
- MMU2 enabled.
- SKR Pro 1.1
- TMC2209
- SENSORLESS HOMING.
- CORE XY

Test 1 :  Setting `EXTRUDER` to `4`
 - HOME X : OK
 - HOME Y : KO

Test 2 :  Setting `EXTRUDER` to `3`
 - HOME X : OK
 - HOME Y : OK

So it's directly linked to the EXTRUDER number.

kursatu, 2021-08-10:
@Minims  I could not replicate your finding. I've set the extruder to 4 or 6 and modified the Sanity check to only check for EXTRUDERS < 2 not for != 5
The problem happens, with the optical endstops. I did not check if it happens with sensorless.
I doubt that there is any replication mistake on my end.
I have very similar HW to yours, as I wrote in the origin issue description.

Minims, 2021-08-10:
Hum, quite strange. I will make a new test with the 10/8 bugfix.
Can it be because I have a `TRIPLE Z` on my CoreXY ? 
Can you try with `EXTRUDER`  = `3` or `2`

Minims, 2021-08-10:
> Hum, quite strange. I will make a new test with the 10/8 bugfix.
> Can it be because I have a `TRIPLE Z` on my CoreXY ?
> Can you try with `EXTRUDER` = `3` or `2`

I have made a new try, I confirm there is no issue on my setup if I use `EXTRUDER = 3`

kursatu, 2021-08-20:
I've just tried it with EXTRUDERS 2 & 3, optical endstops did work.
The EXTRUDERS 4 & 6 does not work.
This is on the latest bugfix branch.