MarlinFirmware/Marlin #22186: [BUG] ===Error: The program size (130726 bytes) is greater than maximum allowed (126976 bytes)

bubulgum, 2021-06-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to build Anycubic Mega Zero 2.0, and have this Err:
RAM:   [===       ]  30.8% (used 5052 bytes from 16384 bytes)
Flash: [===Error: The program size (130726 bytes) is greater than maximum allowed (126976 bytes)
====*** [checkprogsize] Explicit exit, status 1
===]  103.0% (used 130726 bytes from 126976 bytes)
[image]

### Version of Marlin Firmware

2.0.9

### Printer model

Mega Zero 2.0

### Electronics

No

### Additional information & file uploads

`PIO Core Call Error: "Processing melzi (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html\r\nPLATFORM: Atmel AVR (3.3.0) > Sanguino ATmega1284p (16MHz)\r\nHARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 124KB Flash\r\nDEBUG: Current (simavr) On-board (simavr)\r\nPACKAGES: \r\n - framework-arduino-avr 5.1.0 \r\n - toolchain-atmelavr 1.70300.191015 (7.3.0)\r\nConverting Marlin.ino\r\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\r\nLDF Modes: Finder ~ chain, Compatibility ~ soft\r\nFound 6 compatible libraries\r\nScanning dependencies...\r\nDependency Graph\r\n|-- <U8glib-HAL> 0.4.5\r\n| |-- <Wire> 1.0\r\n| |-- <SPI> 1.0\r\n|-- <Wire> 1.0\r\n|-- <SPI> 1.0\r\n|-- <EEPROM> 2.0\r\n|-- <SoftwareSerial> 1.0\r\nBuilding in debug mode\r\nLinking .pio\\build\\melzi\\debug\\firmware.elf\r\n\r\n\n\nc:/users/86185/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: .pio/build/melzi/debug/firmware.elf section `.text' will not fit in region `text'\r\nc:/users/86185/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: region `text' overflowed by 42970 bytes\r\ncollect2.exe: error: ld returned 1 exit status\r\n*** [.pio\\build\\melzi\\debug\\firmware.elf] Error 1\r\n========================= [FAILED] Took 12.34 seconds =========================\r\nEnvironment Status Duration\r\n------------- -------- ------------\r\nmelzi FAILED 00:00:12.339\r\n==================== 1 failed, 0 succeeded in 00:00:12.339 ===================="
`

ellensp, 2021-06-20:
The Sanguino ATmega1284p only has 128K of flash. you have enabled to many things that makes the exceed this harware limit

bubulgum, 2021-06-20:
> The Sanguino ATmega1284p only has 128K of flash. you have enabled to many things that makes the exceed this harware limit

i did not enable anythung, all default

ellensp, 2021-06-20:
you have to use melzi_optiboot_optimized or melzi_optimized

thisiskeithb, 2021-06-20:
As ellensp explained, disable some features to get your firmware to fit.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-20 by thisiskeithb