MarlinFirmware/Marlin #22194: [FR] Probe Assisted Bed Leveling

jimberg98, 2021-06-21:
### Describe the feature you want

I would like to see a bed leveling process where the user turns all of the adjustment screws all of the way in and then enters a command to tell he firmware to probe the bed in each corner and determine which corner is closest to the nozzle. It would then go to each corner that wasn't the closest at that same height as the closest, deploy the probe and wait for the user to loosen the adjustment screw in that corner until the probe is triggered. Once triggered, it would move on to the other two corners. Once this process is done, the bed should be level.  

It would be nice if the user can define the probing locations. It would be optimal if each location was where the adjustment screw was since that's the fulcrum and won't move up and down as much when the other screws are adjusted. It may also be best if when checking the other corners that it moved in order of farthest to closest.

This could be a lot more precise than using a piece of paper and relying on feel.

If this is too complicated, adding a parameter to G30 to tell it to wait for the probe to be triggered would work, too.

ellensp, 2021-06-21:
have you seen https://marlinfw.org/docs/gcode/G035.html

thisiskeithb, 2021-06-21:
> have you seen https://marlinfw.org/docs/gcode/G035.html

Bingo. This feature already exists as `G35`.

closed 2021-06-21 by thisiskeithb

jimberg98, 2021-06-21:
That's good to know. The documentation should be changed to use more obvious vocabulary. Trammiing--. How could I have possibly missed that? ;-)

InsanityAutomation, 2021-06-21:
> 
> 
> That's good to know. The documentation should be changed to use more obvious vocabulary. Trammiing--. How could I have possibly missed that? ;-)

Tramming is the correct term. In actuality, bed leveling is the incorrect term as we are not earth leveling anything. We are tramming the machine to itself. Even the ABL methods in theory should be called compensation, but too much history to change it all now...

jimberg98, 2021-06-22:
> Tramming is the correct term. In actuality, bed leveling is the incorrect term as we are not earth leveling anything. We are tramming the machine to itself. Even the ABL methods in theory should be [...]

Yes, I understand that. The problem is that tramming is a term that is obscure to most people. Since documentation is meant to educate, it's advisable to use the common term, even if technically incorrect, along with the correct term. The reader will then learn the correct term. I had no idea that the feature I was looking for is called tramming. . There's nothing in the explanation of it that matched with what I was searching for. We've all been there looking for answers to questions but can't find the answer because you're not sure of the correct terminology.

I don't think G35 even does what I was asking for. It's close. I did, however, find the 4 corner leveling feature for the LCD menu. That does sort of do what I was asking for. Does anyone know if that can be initiated with a G-code? If not, can a parameter be added to G35 to do what the LCD four corner leveling does?

Thanks.