MarlinFirmware/Marlin #22195: [BUG] (Z multi endstop resulting in errors when compiling)

kurtis-potier-geofabrica, 2021-06-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm compiling Marlin for a machine with a Bigtreetech GTR1.0. I have had success with this is an earlier bugfix version a couple of months ago. I decided to upgrade to the new 2.0.9 version. My machine has 4 Z endstops. When compiling, 
[image]

Thank you!

### Bug Timeline

new

### Expected behavior

I would like to be able to compile this and get some of the new features available in the latest Marlin

### Actual behavior

Compiling failed. My terminal output isn't super helpful but it points to the files M666.cpp, serial.h, macros.h. This line is underlined as the issue in M666.cpp: 
REPEAT_S(2, INCREMENT(NUM_Z_STEPPER_DRIVERS), _ECHO_ZADJ)
with the comment:
expected a ')'C/C++(18)
I've been through this twice to make sure it was nothing I did to trigger it. 

### Steps to Reproduce

1) Enable Z_multi_endstops in configuration.h, select board as BOARD_BTT_GTR_V1_0, baudrate to 115200, serial_port to -1. 
2) Enable X,Y,Z max plugs to use for extra endstops, and Z2,3,4 stepper drivers. Specify motors on the extender pins to be Z2,3,4
3) Set Z stepper drivers to 4 in Configuration_adv.h and enable multiple Z endstops
3)Building will result in errors. 

### Version of Marlin Firmware

2.0.9

### Printer model

Custom

### Electronics

Bigtreetech GTR1.0

### Add-ons

M5 Extender

### Your Slicer

Cura

### Additional information & file uploads

[Configuration + pins.zip](https://github.com/MarlinFirmware/Marlin/files/6689511/Configuration.%2B.pins.zip)

slowbro, 2021-06-22:
Can you copy/paste the whole error into the issue, please?

ellensp, 2021-06-22:
this issue is in  Marlin/src/gcode/calibrate/M666.cpp line 97
`REPEAT_S(2, INCREMENT(NUM_Z_STEPPER_DRIVERS), _ECHO_ZADJ)`
replacing this line with the following works. 
```CPP
        #if NUM_Z_STEPPER_DRIVERS >= 3
          _ECHO_ZADJ(2)
          _ECHO_ZADJ(3)
          #if NUM_Z_STEPPER_DRIVERS >= 4
            _ECHO_ZADJ(4)
          #endif
        #endif
```
but then there are more error with NUM_Z_STEPPER_DRIVERS >- 4 and  Z_MULTI_ENDSTOPS

I have no idea whats going on with the REPEAT_S why it is failing....

ellensp, 2021-06-22:
Second issue is in Marlin/src/module/stepper.cpp

The following doesn't seem to work as intended

```CPP
  if (separate_multi_axis) {                                                                                                \
    if (ENABLED(A##_HOME_TO_MIN)) {                                                                                         \
      if (!(TEST(endstops.state(), A##_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
      if (!(TEST(endstops.state(), A##2_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
      if (!(TEST(endstops.state(), A##3_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##3_motor) A##3_STEP_WRITE(V); \
      if (!(TEST(endstops.state(), A##4_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##4_motor) A##4_STEP_WRITE(V); \
    }                                                                                                                       \
    else {                                                                                                                  \
      if (!(TEST(endstops.state(), A##_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
      if (!(TEST(endstops.state(), A##2_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
      if (!(TEST(endstops.state(), A##3_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##3_motor) A##3_STEP_WRITE(V); \
      if (!(TEST(endstops.state(), A##4_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##4_motor) A##4_STEP_WRITE(V); \
    }                                                                                                                       \
  }                                                                                                                         \
```
replacing it with the following seems to work as expected.
```CPP
if (separate_multi_axis) {                                                                                                \
      if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##_MIN,A##_HOME_TO_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
      if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##2_MIN,A##2_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
      if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##3_MIN,A##3_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##3_motor) A##3_STEP_WRITE(V); \
      if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##4_MIN,A##4_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##4_motor) A##4_STEP_WRITE(V); \
  }                                                                                                                         \

```

ellensp, 2021-06-22:
A diff to make it a bit clearer

```DIFF
+++ b/Marlin/src/gcode/calibrate/M666.cpp
@@ -94,7 +94,14 @@
       #endif
       #if ENABLED(Z_MULTI_ENDSTOPS)
         #define _ECHO_ZADJ(N) SERIAL_ECHOPAIR(" Z" STRINGIFY(N) ":", endstops.z##N##_endstop_adj);
-        REPEAT_S(2, INCREMENT(NUM_Z_STEPPER_DRIVERS), _ECHO_ZADJ)
+        //REPEAT_S(2, INCREMENT(NUM_Z_STEPPER_DRIVERS), _ECHO_ZADJ)
+        #if NUM_Z_STEPPER_DRIVERS >= 3
+          _ECHO_ZADJ(2)
+          _ECHO_ZADJ(3)
+          #if NUM_Z_STEPPER_DRIVERS >= 4
+            _ECHO_ZADJ(4)
+          #endif
+        #endif
       #endif
       SERIAL_EOL();
     }
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 062049ec77..be8d514171 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -316,18 +316,10 @@ xyze_int8_t Stepper::count_direction{0};
 
 #define QUAD_ENDSTOP_APPLY_STEP(A,V)                                                                                        \
   if (separate_multi_axis) {                                                                                                \
-    if (ENABLED(A##_HOME_TO_MIN)) {                                                                                         \
-      if (!(TEST(endstops.state(), A##_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
-      if (!(TEST(endstops.state(), A##2_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
-      if (!(TEST(endstops.state(), A##3_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##3_motor) A##3_STEP_WRITE(V); \
-      if (!(TEST(endstops.state(), A##4_MIN) && count_direction[_AXIS(A)] < 0) && !locked_##A##4_motor) A##4_STEP_WRITE(V); \
-    }                                                                                                                       \
-    else {                                                                                                                  \
-      if (!(TEST(endstops.state(), A##_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
-      if (!(TEST(endstops.state(), A##2_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
-      if (!(TEST(endstops.state(), A##3_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##3_motor) A##3_STEP_WRITE(V); \
-      if (!(TEST(endstops.state(), A##4_MAX) && count_direction[_AXIS(A)] > 0) && !locked_##A##4_motor) A##4_STEP_WRITE(V); \
-    }                                                                                                                       \
+    if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##_MIN,A##_HOME_TO_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##_motor) A##_STEP_WRITE(V);    \
+    if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##2_MIN,A##2_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##2_motor) A##2_STEP_WRITE(V); \
+    if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##3_MIN,A##3_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##3_motor) A##3_STEP_WRITE(V); \
+    if (!(TEST(endstops.state(), (TERN(A##_HOME_TO_MIN,A##4_MIN,A##4_MAX))) && count_direction[_AXIS(A)] < 0) && !locked_##A##4_motor) A##4_STEP_WRITE(V); \
   }                                                                                                                         \
   else {                                                                                                                    \
     A##_STEP_WRITE(V); 
```

kurtis-potier-geofabrica, 2021-06-22:
> 
> 
> Can you copy/paste the whole error into the issue, please? Unless that _is_ the whole thing, in which case, ignore me :)

[error.txt](https://github.com/MarlinFirmware/Marlin/files/6695227/error.txt)
Here is a text file with the terminal output from VScode, thank you for your help!

kurtis-potier-geofabrica, 2021-06-22:
I have made the changes and have it building. I will test on the physical machine today and report on how its doing.

kurtis-potier-geofabrica, 2021-06-22:
I have tested on the machine with exactly the changes suggested. All endstops are working fine! Thank you

ellensp, 2021-06-23:
This needs a pr to fit it in marlin for all.

ellensp, 2021-06-27:
This PR has been merged, closing

closed 2021-06-27 by ellensp