MarlinFirmware/Marlin #22197: [BUG] Ender 3 Custom Bootscreen not showing on BTT SKR Mini E3 v2.0

Ffestiniog, 2021-06-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have been trying to get the Ender bootscreen to show on my Ender 3 after replacing the board with a bigtreetech SKR Mini E3 mainboard, I have enabled SHOW_CUSTOM_BOOTSCREEN in the firmware and am using the example configuration files for this board and printer. However it is not showing the bootscreen, it just shows a blank screen for the amount of time specified in the _Bootscreen.h file and then the marlin bootscreen.

### Bug Timeline

Started when I replaced the board

### Expected behavior

I expected it to show the Ender bootscreen

### Actual behavior

It showed a blank screen

### Steps to Reproduce

1. Set up the configuration files as attached
2. Compile the firmware
3. turn on the printer

### Version of Marlin Firmware

2.0.8.2

### Printer model

Creality Ender 3

### Electronics

BTT SKR mini E3 upgrade board

### Your Slicer

Cura

### Additional information & file uploads

Log output:
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
echo:M105
echo:M105
echo:M105
Just keeps echoing M105

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/6693497/Configuration.files.zip)

ellensp, 2021-06-22:
Config files not bugfix, SHOW_CUSTOM_BOOTSCREEN is disabled and no _Bootscreen.h attached

Have updated your configs to bugfix, try these
[bugfix_configs.zip](https://github.com/MarlinFirmware/Marlin/files/6693859/bugfix_configs.zip)

Ffestiniog, 2021-06-22:
> Config files not bugfix, SHOW_CUSTOM_BOOTSCREEN is disabled and no _Bootscreen.h attached
> 
> Have updated your configs to bugfix, try these
> [bugfix_configs.zip](https://github.com/MarlinFirmware/Marlin/files/6693859/bugfix_configs.zip)

Sorry think I put wrong config version on here, forgot bootscreen.h! Will try those configs!

thisiskeithb, 2021-06-22:
You probably forgot to copy all the correct files/configs because it works fine here.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-22 by thisiskeithb