MarlinFirmware/Marlin #22198: [BUG] (Mixing Extruder not working stepper E1)

nayim12, 2021-06-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin bugfiz 2.0.x 
Configuration H Version 02000900
When i select Mixing extruder with 2 mixing steppers….
 only 1 stepper is running and othe one has no signal
Bt when i select number of extruder 2  and single nozzle .. both steppers are working

bt in my old FW of marlin bugfix 2.0.x
Configuration H Version 020007
 Everything work poperly

when i update my fw 020007 to 020009 the bug appairs
[image]
[image]

### Expected behavior

Run 2 steppers 

### Actual behavior

Running 1 steppers

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Electronics

Atmega 2560+ramps 1.4

ellensp, 2021-06-22:
suspect it might be related to #define DISABLE_INACTIVE_EXTRUDER

nayim12, 2021-06-22:
suspect it might be related to #define DISABLE_INACTIVE_EXTRUDER
i was try bt no result

nayim12, 2021-06-22:
[image]

ellensp, 2021-06-22:
Please attach your configuration files.  not jpegs of parts of it

nayim12, 2021-06-22:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02000900

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 * Calibration Objects: https://www.thingiverse.com/thing:5573
 *                      https://www.thingiverse.com/thing:1278865
 */

//===========================================================================
//========================== DELTA / SCARA / TPARA ==========================
//===========================================================================
//
// Download configurations from the link above and customize for your machine.
// Examples are located in config/examples/delta, .../SCARA, and .../TPARA.
//
//===========================================================================

// @section info

[... 2779 lines cut ...]
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

ellensp, 2021-06-22:
As you see pasting as teXt makes a mess of things, and make it unusable.  
Please zip up your Configuration.h and Configuration_adv.h into a zip called for eg configs.zip and attach this zip file  
Then detete the pasted version.

nayim12, 2021-06-22:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6693945/config.zip)

ellensp, 2021-06-22:
Now please fill in in the 
Steps to Reproduce
Presume I've never used a mixing extruder.

DerAndere1, 2021-06-22:
This could be a duplicate of issue https://github.com/MarlinFirmware/Marlin/issues/22166

nayim12, 2021-06-22:
can i get the solution

ellensp, 2021-06-22:
@nayim12 no one knows yet, that is why it is still open

closed 2021-06-22 by nayim12