MarlinFirmware/Marlin #22200: [BUG] PS_ON_PIN (12) not present on MKS Gen-L

HoverClub, 2021-06-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

PS_ON_PIN defined as D12 in src/pins/pins_RAMPS.h (included by src/pins/pins_MKS_GEN_L..h) does not exist on the official MKS Gen L board (https://github.com/makerbase-mks/MKS-GEN_L - it's not available on any connector) - this is also true for some derivative RAMPS boards.  Possible solution would be to use the 4th Servo signal pin (D11)? 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6695661/config.zip)

### Bug Timeline

Today

### Version of Marlin Firmware

2.09

ellensp, 2021-06-22:
is already in a #ifdef block
```
#ifndef PS_ON_PIN
  #define PS_ON_PIN                           12
#endif

```

All you need to do is #define PS_ON_PIN 11  in your pins file or in Configuration.h  and it overrides this pin.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2021-06-22:
The board does not have a pin allocated to this function. So it drops to the default for the ramps.
If you add a power supply of this type, you are expected to know where you plugged it in and set PS_ON_PIN accordingly

HoverClub, 2021-06-22:
It would be much more helpful if the MKS pin file generated a compile error to remind users that pin12 doesn't exist (or needs to be allocated) rather than defaulting to a non-existent pin?  Certainly would have saved me an hour or so trying to figure out which pin to connect the PSU to!
```
#ifndef PS_ON_PIN
  #error "MKS GEN L doesn't support PS_ON_PIN on RAMPS pin 12 - set PS_ON_PIN to another pin in configuration.h
#endif

```