MarlinFirmware/Marlin #22201: [BUG] (bug summary)

niallfo, 2021-06-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

** Assisted Tramming Wizard ** BL Touch V3.1** LCD function to display measurement deviations has a bug. Have been trying multiple firmware versions for quite some time, bug fixes and none bugfixes all have the same problem. When sending probe to "Left-Front" on LCD screen and probing for the measurement, it does not report back the measurement deviation to the LCD screen. Instead, it displays 0 until I switch to a different corner and before taking that new measurement, seeing the last measurement in the LCD screen, then switch back to re-probe, and back and forth etc. to level that specific corner "Front-Left". Have no problems with any of the other corners just always that corner. Also have to spin my adjustment knob in the opposite direction in comparison to the others to level according to the last deviation reported for 'Front-Left'.

### Bug Timeline

No, this has always been happening 

### Version of Marlin Firmware

2.0.x

### Printer model

ender 5

### Add-ons

BL Touch V3.1

### Your Slicer

Cura

niallfo, 2021-06-23:
The board I am using with the ender 5 is v1.1.1.4

niallfo, 2021-06-26:
So it seems tht it is only "#define TRAMMING_POINT_NAME_1" that doesn't report measurement deviations to LCD screen after probing. I just had to add an extra tramming point (TRAMMING_POINT_NAME_5) to that corner to solve the problem. 

// Define positions for probe points.
  #define TRAMMING_POINT_XY { {  174, 174 }, { 46,  174 }, { 46, 46 }, { 174, 46}, {  174, 174 } }

  // Define position names for probe points.
  #define TRAMMING_POINT_NAME_1 "Do Not Use"
  #define TRAMMING_POINT_NAME_2 "Back-Left"
  #define TRAMMING_POINT_NAME_3 "Front-Left"
  #define TRAMMING_POINT_NAME_4 "Front-Right"
  #define TRAMMING_POINT_NAME_5 "Back-Right"

  #define RESTORE_LEVELING_AFTER_G35    // Enable to restore leveling setup after operation
  #define REPORT_TRAMMING_MM          // Report Z deviation (mm) for each point relative to the first

  #define ASSISTED_TRAMMING_WIZARD    // Add a Tramming Wizard to the LCD menu

thinkyhead, 2021-07-30:
It might be by design. From my quick examination of the code it appears that the Left-Front corner is used as the base value and all the other corners are adjusted relative to it.

niallfo, 2021-08-02:
No it doesn't display a value after you probe that corner ever.  I don't use tramming point 1 anymore and just made tramming point 5 the extra cornerand it still works. It probes the center first which i assumed was what all other corners were relative too after the initial center probe done at the start. I think I have been leveling the bed just fine? or does that not make sense with the code?

niallfo, 2021-08-19:
That 1 corner does not change values no matter what adjustment is made. Im curious why its there if it is only the starting reference. I believe your looking at the "corner bed leveling" tramming function which moves to each corner to level based off the first corner probed. I am talking about the tramming wizard which starts with a probe in the middle and then manually select four the four outer corners.

niallfo, 2021-08-19:
> 
> 
> It might be by design. From my quick examination of the code it appears that the Left-Front corner is used as the base value and all the other corners are adjusted relative to it. Left-Front itself is not adjusted.

No  it doesn't display a value after you  probe that corner ever. I don't use tramming point 1 anymore and just made tramming point 5 the extra  cornerand it still works. It  probes the center first which i assumed was what all other corners were relative too after the initial center probe done at the start. I think I have been leveling the bed just fine? or does that not make sense with the code

niallfo, 2021-08-19:
> 
> 
> It might be by design. From my quick examination of the code it appears that the Left-Front corner is used as the base value and all the other corners are adjusted relative to it. Left-Front itself is not adjusted.

 That 1 corner does not change values no matter what adjustment is made. Im curious why its there if it is only the starting reference. I believe your looking at the "corner bed leveling" tramming function which moves to each corner to level based off the first corner probed. I am talking about the tramming wizard which starts with a probe in the middle and then manually select four the four outer corners.

thinkyhead, 2021-08-20:
Who is the original author of that tramming wizard…? @qwewer0 Maybe we can bring them in to devise a patch.

qwewer0, 2021-08-20:
**G35** is added by @rmcbc (#16897), and **ASSISTED_TRAMMING_WIZARD** by @Speaka (#20000)

It displays the "correct" value for the first point, as every measurement is relative to the first point and not the firstly probed point:
https://github.com/MarlinFirmware/Marlin/blob/193205a40dd6b756bd69cab6fcb1af63ddd6d452/Marlin/src/lcd/menu/menu_tramming.cpp#L65

Every time the first point is probed the relative value is also updated:

https://github.com/MarlinFirmware/Marlin/blob/193205a40dd6b756bd69cab6fcb1af63ddd6d452/Marlin/src/lcd/menu/menu_tramming.cpp#L54

We could save the first value as a base, (something better(?) like this:)

```
	float z_measured[G35_PROBE_COUNT] = { 0 };
->	float z_base;
->	bool is_first = true;
	static uint8_t tram_index = 0;
	
	#if HAS_LEVELING
	#include "../../feature/bedlevel/bedlevel.h"
	#endif
	
	static bool probe_single_point() {
	do_blocking_move_to_z(TERN(BLTOUCH, Z_CLEARANCE_DEPLOY_PROBE, Z_CLEARANCE_BETWEEN_PROBES));
	// Stow after each point with BLTouch "HIGH SPEED" mode for push-pin safety
	const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[tram_index], TERN(BLTOUCH_HS_MODE, PROBE_PT_STOW, PROBE_PT_RAISE), 0, true);
	DEBUG_ECHOLNPAIR("probe_single_point: ", z_probed_height, "mm");
->	if (is_first) {
		z_base = z_probed_height;
		is_first = false;
	}
	z_measured[tram_index] = z_probed_height;
	move_to_tramming_wait_pos();
	
	return !isnan(z_probed_height);
	}
	
	static void _menu_single_probe(const uint8_t point) {
	tram_index = point;
	DEBUG_ECHOLNPAIR("Screen: single probe screen Arg:", point);
	START_MENU();
	STATIC_ITEM(MSG_BED_TRAMMING, SS_LEFT);
->	STATIC_ITEM(MSG_LAST_VALUE_SP, SS_LEFT, ftostr42_52(z_base - z_measured[point])); // Print diff
```

Speaka, 2021-08-20:
@thinkyhead  is right, The first probing is base. From here the measurements in any other corner are displayed as relative values. Therefore the value in the first corner is always 0 by design as theres no difference to the base value.
In other Words: The bed is levelled to the height of the first corner. And the wizard shows the deviation of the other corners in comparison to the first.

sjasonsmith, 2021-08-28:
I agree that the current behavior is by design, but that it would be more intuitive if any point could be selected as the "first" point.

I'm about to post a PR to fix another annoyance in this wizard. I'm already messing with the z_measured reporting, so I can easily make that change at the same time.

niallfo, 2021-08-31:
> @thinkyhead and @qwewer0 are right, The first probing is the base level. From here the measurements in any other corner are displayed as relative values. Therefore the value in the first corner is always 0 by design as theres no difference to the base value.
> In other Words: The bed is levelled to the height of the first corner. And the wizard shows the deviation of the other corners in comparison to the first.

No but the first corner I just to probe on the LCD screen never shows 0

niallfo, 2021-09-01:
> **G35** is added by @rmcbc (#16897), and **ASSISTED_TRAMMING_WIZARD** by @Speaka (#20000)
>
> It displays the "correct" value for the first point, as every measurement is relative to the first point [...]

https://user-images.githubusercontent.com/86325524/131593532-387ab956-9bc2-444e-9c86-5615826d28e2.mp4

thisiskeithb, 2021-09-15:
Resolved in https://github.com/MarlinFirmware/Marlin/pull/22657

closed 2021-09-15 by thisiskeithb