MarlinFirmware/Marlin #22202: [PROBLEM] (Ender-3 v2 with 4.2.2 board)

zonalimitatore, 2021-06-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With the last release on a ender 3 v2 with 4.2.2 silent board,i can't compile it due this problem on compile.
If i try to compile with stock cr10 screen, i have black screen and beeeeeep

Marlin\src\lcd\dwin\dwin_lcd.cpp: In function 'void DWIN_Send(size_t&)':
Marlin\src\lcd\dwin\dwin_lcd.cpp:85:20: error: 'MYSERIAL1' was not declared in this scope
   LOOP_L_N(n, i) { MYSERIAL1.write(DWIN_SendBuf[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
Marlin\src\lcd\dwin\dwin_lcd.cpp:85:20: note: suggested alternative: 'MYSERIAL0'
   LOOP_L_N(n, i) { MYSERIAL1.write(DWIN_SendBuf[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
                    MYSERIAL0
Marlin\src\lcd\dwin\dwin_lcd.cpp:86:20: error: 'MYSERIAL1' was not declared in this scope
   LOOP_L_N(n, 4) { MYSERIAL1.write(DWIN_BufTail[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
Marlin\src\lcd\dwin\dwin_lcd.cpp:86:20: note: suggested alternative: 'MYSERIAL0'
   LOOP_L_N(n, 4) { MYSERIAL1.write(DWIN_BufTail[n]); delayMicroseconds(1); }
                    ^~~~~~~~~
                    MYSERIAL0
Marlin\src\lcd\dwin\dwin_lcd.cpp: In function 'bool DWIN_Handshake()':
Marlin\src\lcd\dwin\dwin_lcd.cpp:97:10: error: 'MYSERIAL1' was not declared in this scope
   while (MYSERIAL1.available() > 0 && recnum < (signed)sizeof(databuf)) {
          ^~~~~~~~~
Marlin\src\lcd\dwin\dwin_lcd.cpp:97:10: note: suggested alternative: 'MYSERIAL0'
   while (MYSERIAL1.available() > 0 && recnum < (signed)sizeof(databuf)) {
          ^~~~~~~~~
          MYSERIAL0
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\marlinui.cpp.o
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\dwin\dwin_lcd.cpp.o] Error 1

### Version of Marlin Firmware

2.0.9

### Printer model

ender 3 v2

### Electronics

creality v 4.2.2 silent

ellensp, 2021-06-23:
Cannot replicate this issue
Delete your hidden `Marlin/.pio` directory and try again

ellensp, 2021-06-23:
I also know that @CRCinAU compiles for printer and board daily. If there where a compiling issue he would be all over it.

CRCinAU, 2021-06-23:
Yep - no issues detected here... Ensure you're using the latest sets of Configuration files from here:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2

CRCinAU, 2021-06-23:
At the same time though, I don't know why you're trying to enable a CR10 screen on the V2 unless you've replaced the screen.

ellensp, 2021-06-23:
disabling DWIN_CREALITY_LCD and enabling CR10_STOCKDISPLAY and adding to required #define RET6_12864_LCD
that marlin tells you about. Also compiles without out issues.

zonalimitatore, 2021-06-23:
Ok.I try the solutions

zonalimitatore, 2021-06-23:
With clean version of marlin and last config,i've got this:

`In file included from C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30:0,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\shared\marduino.h:36,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\stm32f1\hal.h:32,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\hal.h:30,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\inc\marlinconfig.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../compat.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\flash_storage.cpp:23:
Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../../ui_api.h: In function 'void ExtUI::simulateUserClick()':
C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:86:14: error: assignment of read-only variable 'MarlinUI::lcd_clicked'
 #define true 0x1
              ^
Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../../ui_api.h:193:54: note: in expansion of macro 'true'
   inline void simulateUserClick() { ui.lcd_clicked = true; }
                                                      ^~~~
In file included from C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30:0,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\shared\marduino.h:36,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\stm32f1\hal.h:32,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\hal.h:30,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\inc\marlinconfig.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../compat.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\media_file_reader.cpp:23:
Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../../ui_api.h: In function 'void ExtUI::simulateUserClick()':
C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:86:14: error: assignment of read-only variable 'MarlinUI::lcd_clicked'
 #define true 0x1
              ^
Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\../../ui_api.h:193:54: note: in expansion of macro 'true'
   inline void simulateUserClick() { ui.lcd_clicked = true; }
                                                      ^~~~
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\flash_storage.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\media_file_reader.cpp.o] Error 1
In file included from C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30:0,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\shared\marduino.h:36,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\stm32f1\hal.h:32,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\hal.h:30,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\inc\marlinconfig.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../compat.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../config.h:24,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\confirm_home_e.cpp:23:
Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../../ui_api.h: In function 'void ExtUI::simulateUserClick()':
C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:86:14: error: assignment of read-only variable 'MarlinUI::lcd_clicked'
 #define true 0x1
              ^
Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../../ui_api.h:193:54: note: in expansion of macro 'true'
   inline void simulateUserClick() { ui.lcd_clicked = true; }
                                                      ^~~~
In file included from C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30:0,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\shared\marduino.h:36,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\stm32f1\hal.h:32,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\hal.h:30,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\inc\marlinconfig.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../compat.h:31,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../config.h:24,
                 from Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\confirm_home_xyz.cpp:23:
Marlin\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\../../ui_api.h: In function 'void ExtUI::simulateUserClick()':
[... cut ...]
================================================================================================= [FAILED] Took 44.00 seconds =================================================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:44.004
============================================================================================ 1 failed, 0 succeeded in 00:00:44.004 ============================================================================================`

ellensp, 2021-06-23:
current marlin build env STM32F103RET6_creality does not use maple and your getting framework-arduinoststm32-maple errors.

delete .pio directory in Marlin

ellensp, 2021-06-23:
marlin 2.0.8.2 is the last version where STM32F103RET6_creality uses maple.
2.0.9 created new build environment STM32F103RET6_creality_maple wich uses maple and STM32F103RET6_creality wich does not

If you upgrade from 2.0.8.2 or older to 2.0.9 or newer. you have to remove the .pio directory as platformio does not realize that the previous STM32F103RET6_creality is not the same as the newer STM32F103RET6_creality

The-EG, 2021-06-23:
> disabling DWIN_CREALITY_LCD and enabling CR10_STOCKDISPLAY and adding to required #define RET6_12864_LCD
that marlin tells you about. Also compiles without out issues. (but I haven't tested actually works)

I can confirm that this does actually work if you replace the DWIN with the 'old' LCD (same as a E3Pro or v1). I've been using that setup on my Ender 3 v2 for several months. After swapping the screens out, it's essentially an Ender 3 Pro with a 4.2.2 board and you could just use that example as a starting point instead.

zonalimitatore, 2021-06-23:
Yes.I've do it. But same errors as before.
This is when i've clean the folder and rebuild with a clean config:
`Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/framework-arduinoststm32-maple @ ~3.10000.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoststm32-maple @ 3.10000.201129 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2`

And again the same error

ellensp, 2021-06-23:
still says framework-arduinoststm32-maple 
still an old marlin/platfomio.ini not current release or bugfix

ellensp, 2021-06-23:
should say  for STM32F103RET6_creality
```
Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
```

or if STM32F103RET6_creality_maple 
```
Processing STM32F103RET6_creality_maple (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
```

ellensp, 2021-06-23:
delete your `C:\Users\Bax\.platformio` force platformio to reinstall all build environments and other dependencies

zonalimitatore, 2021-06-23:
ok i try

ellensp, 2021-06-23:
check your ini/stm32f1.ini files

It should have 

```
[common_stm32]
platform      = ststm32@~12.1
```

not 12.0

or if using maple ini/stm32f1-maple.ini

should have

```
[common_stm32f1]
platform          = ststm32@~12.1

still not 12.0
```

zonalimitatore, 2021-06-23:
DAMN....The same...Continue to give me maple:

`Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.70201.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\i2c_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\iwdg.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\nvic.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\pwr.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\rcc.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\rcc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\spi.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\spi_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\stm32f1\performance\isrs.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\stm32f1\performance\vector_table.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\systick.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\timer.c.o
[... 329 lines cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\ftdi_eve_touch_ui\bioprinter\status_screen.cpp.o] Error 1
In file included from C:\Users\Bax\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30:0,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\shared\marduino.h:36,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\stm32f1\hal.h:32,
                 from c:\users\bax\desktop\happygross - copia\marlin\src\hal\hal.h:30,
[... cut ...] ========================
Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:07:31.319
============================================================================================ 1 failed, 0 succeeded in 00:07:31.319 ============================================================================================ 
The terminal process "C:\Users\Bax\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
`

zonalimitatore, 2021-06-23:
Now,deleting 
board_build.core  = maple

`Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.90201.191206 has been installed!
Tool Manager: Installing platformio/framework-cmsis @ ~2.50501.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-cmsis @ 2.50501.200527 has been installed!
Tool Manager: Installing platformio/framework-arduinoststm32 @ ~4.10900.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoststm32 @ 4.10900.200819 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <EEPROM> 2.0.1
|-- <Wire> 1.0
|-- <Servo> 1.1.2
|-- <IWatchdog> 1.0.0
Building in release mode`

`In file included from Marlin\src\HAL\STM32F1\tft\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:23:
Marlin\src\HAL\STM32F1\tft\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\shared\Delay.cpp:24:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\shared\servo.cpp:53:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from Marlin\src\HAL\STM32F1\tft\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\tft\xpt2046.cpp:23:
Marlin\src\HAL\STM32F1\tft\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
[... 64 lines cut ...]
                 from Marlin\src\feature\../inc/../HAL/HAL.h:30,
                 from Marlin\src\feature\../inc/MarlinConfig.h:31,
                 from Marlin\src\feature\ammeter.cpp:23:
Marlin\src\feature\../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RET6_creality\src\src\feature\ammeter.cpp.o] Error 1
In file included from Marlin\src\feature\../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\feature\../inc/../HAL/HAL.h:30,
                 from Marlin\src\feature\../inc/MarlinConfig.h:31,
                 from Marlin\src\feature\babystep.cpp:23:
Marlin\src\feature\../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RET6_creality\src\src\feature\babystep.cpp.o] Error 1
================================================================================================= [FAILED] Took 562.88 seconds =================================================================================================
Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:09:22.880
============================================================================================ 1 failed, 0 succeeded in 00:09:22.880 ============================================================================================ 
The terminal process "C:\Users\Bax\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

ellensp, 2021-06-23:
There is a lot of code in that log that platformio should be skipping as not needed in there, that is is not skipping.

for eg my full log doesn't include the string ftdi_eve_touch_ui at all, it is skipped.

Start from scratch

you downloaded marlin
you downloaded the example configs
you copy the example you want into marlin.

now you open vscode 
now go to the menu.  File | Open Folder
select the marlin directory with the platformio.ini file in it.

now build. 
If using the marlin auto builder you select a build environment from the menu
If building using the build icon on bottom task bar ov vscode you first need to set default_envs in platfomio.ini to 
either STM32F103RET6_creality or STM32F103RET6_creality_maple.

This is my full build log
[out.zip](https://github.com/MarlinFirmware/Marlin/files/6701734/out.zip)

zonalimitatore, 2021-06-23:
> 
> 
> There is a lot of code in that log that platformio should be skipping as not needed in there, that it is not skipping.
> 
> for eg my full log doesn't include the string ftdi_eve_touch_ui at all, it is skipped.
> 
> Start from scratch
> 
> you downloaded marlin
> you downloaded the example configs
> you copy the example you want into marlin.
> 
> now you open vscode
> now go to the menu. File | Open Folder
> select the marlin directory with the platformio.ini file in it.
> 
> now build.
> If using the marlin auto builder you select a build environment from the menu
> If building using the build icon on bottom task bar ov vscode you first need to set default_envs in platfomio.ini to
> either STM32F103RET6_creality or STM32F103RET6_creality_maple.
> 
> This is my full build log
> [out.zip](https://github.com/MarlinFirmware/Marlin/files/6701734/out.zip)

Is what i've done,man :(

ellensp, 2021-06-23:
make clean and delete .pio filder and zip it up and attach it for examination

zonalimitatore, 2021-06-23:
ok

ellensp, 2021-06-23:
hopefully its small enough to allow that....

zonalimitatore, 2021-06-23:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6701768/Marlin.zip)

Here attached!

ellensp, 2021-06-23:
one directory to high, want to see platformio.ini and buildroot directory

thisiskeithb, 2021-06-23:
<details><summary>Your configs build fine:</summary>
<p>

```prolog
Indexing .pio/build/STM32F103RET6_creality/lib719/libWire.a
Linking .pio/build/STM32F103RET6_creality/firmware-20210623-053633.elf
Checking size .pio/build/STM32F103RET6_creality/firmware-20210623-053633.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.1% (used 7304 bytes from 65536 bytes)
Flash: [===       ]  25.9% (used 135832 bytes from 524288 bytes)
Building .pio/build/STM32F103RET6_creality/firmware-20210623-053633.bin
======================================================= [SUCCESS] Took 31.59 seconds =======================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  SUCCESS   00:00:31.586
======================================================== 1 succeeded in 00:00:31.586 ========================================================

Terminal will be reused by tasks, press any key to close it.
```

</p>
</details>

zonalimitatore, 2021-06-23:
> 
> Your configs build fine:
> 
> ```prolog
> Indexing .pio/build/STM32F103RET6_creality/lib719/libWire.a
> Linking .pio/build/STM32F103RET6_creality/firmware-20210623-053633.elf
> Checking size .pio/build/STM32F103RET6_creality/firmware-20210623-053633.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [=         ]  11.1% (used 7304 bytes from 65536 bytes)
> Flash: [===       ]  25.9% (used 135832 bytes from 524288 bytes)
> Building .pio/build/STM32F103RET6_creality/firmware-20210623-053633.bin
> ======================================================= [SUCCESS] Took 31.59 seconds =======================================================
> 
> Environment             Status    Duration
> ----------------------  --------  ------------
> STM32F103RET6_creality  SUCCESS   00:00:31.586
> ======================================================== 1 succeeded in 00:00:31.586 ========================================================
> 
> Terminal will be reused by tasks, press any key to close it.
> ```

what the ........

ellensp, 2021-06-23:
and there are no odd changes in the marlin source provided.

zonalimitatore, 2021-06-23:
[happygross - Copia.zip](https://github.com/MarlinFirmware/Marlin/files/6701826/happygross.-.Copia.zip)

ellensp, 2021-06-23:
The provided zip there are 55 changed files from marlin 2.0.9 and yes it errors

ellensp, 2021-06-23:
Your platfomio.ini is quite broken and causing most f the issues

it is missing file filters
```

  -<src/lcd/extui/anycubic_chiron>
  -<src/lcd/extui/anycubic_i3mega>
  -<src/lcd/extui/dgus> -<src/lcd/extui/dgus/fysetc> -<src/lcd/extui/dgus/hiprecy> -<src/lcd/extui/dgus/mks> -<src/lcd/extui/dgus/origin>
  -<src/lcd/extui/example>
  -<src/lcd/extui/ftdi_eve_touch_ui>
  -<src/lcd/extui/malyan>
  -<src/lcd/extui/mks_ui>
  -<src/lcd/extui/nextion>

```
and there are lots of other issues

Download a fresh marlin 2.0.9 and replace all files, not just the files from the  Marlin sub directory.

thisiskeithb, 2021-06-23:
Closing since this isn't a Marlin bug.

closed 2021-06-23 by thisiskeithb

zonalimitatore, 2021-06-23:
Now it's working.Thank u guys