MarlinFirmware/Marlin #22204: [BUG] M150 color command in print file has 15 second delay

tjthejuggler, 2021-06-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When i put an M150 color command at the beginning of the print text file, and then İ select the print file and run it, the LEDs dont change colors for 15 seconds. This only happens the first time I print after turning the device on.

### Bug Timeline

New

### Expected behavior

İ expect the LED M150 command to run immediately when i confirm the print.

### Actual behavior

Thee is a 15 second delay before the color command is sent.

### Steps to Reproduce

Put a M150 color command in the text file that will be printed
Power on device
Select the print file
Confirm the print

### Version of Marlin Firmware

Most recent

### Printer model

Ender 3

### Electronics

Mks

ellensp, 2021-06-23:
Please attach your configuration files as "Electronics: MKS" is insufficient they have 20+ different control boards.

tjthejuggler, 2021-06-23:
Thanks so much! Is this what you mean by configuration files? I'm not sure if I understand... now I'm think this might not be right. Where might I find config files specific to the board? Thanks again!

we are using makerbase Mks sgen_l v2

[config_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/6700284/config_adv.h.txt)
[config.h.txt](https://github.com/MarlinFirmware/Marlin/files/6700287/config.h.txt)

ellensp, 2021-06-24:
got an small example gcode you can share with the added M150?

hsrmt, 2021-06-28:
Hey Ellen, here ıs a sample of the gcode file we are using. Thanks so much for the help!

[sample_gcode.txt](https://github.com/MarlinFirmware/Marlin/files/6725167/sample_gcode.txt)

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2021-07-30:
You might try inserting `M118 message` lines in-between each of the starting lines of your G-code file to see where the delay(s) are occurring. If you are using `PRINTER_EVENT_LEDS` try disabling that for some tests.