MarlinFirmware/Marlin #22208: [BUG] (Touch screen stopped working)

lclrcarvalho, 2021-06-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The touchscreen stopped working.  Last version it still works is 2.8.0.2

Tested with most recent bugfix and still happens.  

Usually I leave TOUCH_SCREEN_CALIBRATION commented, but uncommenting it just leaves the screen in calibration mode without reaction to touches.

I'm using 
MKS_ROBIN_TFT43
TFT_COLOR_UI

MKS_ROBIN_NANO35 environment

### Bug Timeline

after 2.8.0.2

### Expected behavior

Have marlin react to touches in the TFT

### Actual behavior

No reaction to touch

### Steps to Reproduce

1. Turn on
2. touch screen
3. nothing happens

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x

### Printer model

TWO TREES BLUER PLUS

### Electronics

MKS Robin Nano V1.2

### Add-ons

TFT with 480x272

### Additional information & file uploads

Serial works, but it takes a simple G28 command takes the screen to the "Nozzle" screen and it's impossible to come out as touch doesn't work.

ellensp, 2021-06-23:
Please attach your configuration files.

lclrcarvalho, 2021-06-23:
[configuration files for bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/6702814/configuration.files.for.bugfix.zip)

Here they are

ellensp, 2021-06-23:
@rhapsodyv could you take a look at this?

rhapsodyv, 2021-06-23:
Can you try enable: 
```
#define TOUCH_SCREEN_CALIBRATION
```
and remove or comment (disable) this values: 
```
  #define TOUCH_CALIBRATION_X 	17653
  #define TOUCH_CALIBRATION_Y 	11931
  #define TOUCH_OFFSET_X 		-51
  #define TOUCH_OFFSET_Y 		-51
  #define TOUCH_ORIENTATION   	TOUCH_PORTRAIT
```

and test again?

lclrcarvalho, 2021-06-24:
Hi,

After booting it goes to the calibration screen in "bottom left" and regardless where I touch it doesn't react at all.

I've also tried via octoprint sending a G28 command. It executes the command fine, but in the screen it flicks and ends up in "Top Left" with 2 plus signs (one in the top left and one in the bottom right).

So as I had tried before just uncommenting TOUCH_SCREEN_CALIBRATION, commenting my calibration settings doesn't cause any different behavior.

Maybe completely unrelated but while compiling I get a warning:

"In file included from Marlin\src\HAL\STM32\eeprom_flash.cpp:33:
C:\Users\Luis\.platformio\packages\framework-arduinoststm32\libraries\EEPROM\src/EEPROM.h:247:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
  247 | static EEPROMClass EEPROM;
      |                    ^~~~~~"

rhapsodyv, 2021-06-24:
if you use the env `mks_robin_nano35_maple`, do it works?

lclrcarvalho, 2021-06-24:
I get the error in the beginning:

*** [C:\Users\Luis\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103v\board.cpp] C:\Users\Luis\Documents\GitHub\Marlin\.pio\build\mks_robin_nano35_maple\.sconsign37.dblite: No such file or directory

So I can't build it...   (I've tried deleting the .pio folder for a fresh compile)

rhapsodyv, 2021-06-24:
just try again, these errors are temporary

lclrcarvalho, 2021-06-24:
The touch screen now works :)

But after pressing the 4 corners (starting on "top left") it ends with the "Top left" screen with to plus signs and it hangs. This is something that happened to me in the past (like back in January), but only sometimes. Now happened consistently  (at least it failed in 5 consecutive starts).

Compiling with my previous calibration figures, takes me to the menu and all seems fine now.

Out of curiosity as I couldn't find anything about the "maple" flavor, what's the difference ?

rhapsodyv, 2021-06-24:
Thanks for the details. I will take a deeper look.

If it is working with _maple, but not with the other env, it seems some spi default config difference between these two envs.

The hang could be related with the auto save and eeprom settings

rhapsodyv, 2021-06-24:
Can you test the env without _maple, but edit this line:

https://github.com/MarlinFirmware/Marlin/blob/4f8191b4818b97bd20eb9db2042dc07c97cce6cc/Marlin/src/HAL/STM32/tft/xpt2046.cpp#L56

Change it from:

SPIx.Init.BaudRatePrescaler  = SPI_BAUDRATEPRESCALER_8;

To:

SPIx.Init.BaudRatePrescaler  = SPI_BAUDRATEPRESCALER_64;

And test again?

lclrcarvalho, 2021-06-25:
Yes, that fixes the touch screen issue.  All seems ok.

Still if I turn on TOUCH_SCREEN_CALIBRATION the same behavior happens (as the maple version).  

Should I create a different bug for that ?

rhapsodyv, 2021-06-25:
Perfect! That confirm my suspect that the issue is related with spi speed defaults.

I will send a PR later.

You don’t need open a new issue for the touch calibration hang. I will take a look at it soon.

It hangs if you disable auto save?

lclrcarvalho, 2021-06-25:
You are right, commenting 
  //#if BOTH(TOUCH_SCREEN_CALIBRATION, EEPROM_SETTINGS)
  //  #define TOUCH_CALIBRATION_AUTO_SAVE // Auto save successful calibration values to EEPROM
  //#endif

Solves the issue.  

After calibration goes into "Calibration Completed" and touching moves to the main screen.

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

rubyredrick, 2021-10-21:
I think I'm seeing a similar issue.

Building for the Elegoo Neptune 2.

I get through the calibration, but then the main screen comes up and there is no response to touches.

For now I've gone back to using the TFT_COLOR_UI 
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/7393419/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/7393420/Configuration_adv.h.txt)

just-trey, 2022-01-13:
@rub you may want to look at what I submitted. It is currenlty in the import branch. This works but be sure to look at the [readme](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Elegoo/Neptune%202#elegoo-neptune-2neptune-2dneptune-2s-configuration)

Bjohnson131, 2022-04-24:
I have just seen this too, tronXY x5sa pro. just downloaded 2.0.9 today.

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/8548951/Configs.zip)

romass, 2022-12-04:
Any updates on this thread?
I have identical issue.
After doing calibration Top L, Bottom L, Top R, Bottom R, it goes to the same state and freezes.

Bjohnson131, 2022-12-04:
> Any updates on this thread? I have identical issue. After doing calibration Top L, Bottom L, Top R, Bottom R, it goes to the same state and freezes.
> 
> My config:
> 
> ```
> #define TOUCH_SCREEN
> #if ENABLED(TOUCH_SCREEN)
>   #define BUTTON_DELAY_EDIT  50 // (ms) Button repeat delay for edit screens
>   #define BUTTON_DELAY_MENU 250 // (ms) Button repeat delay for menus
> 
>   //#define TOUCH_IDLE_SLEEP 300 // (s) Turn off the TFT backlight if set (5mn)
> 
>   #define TOUCH_SCREEN_CALIBRATION
> 
>   // XPT2046 Touch Screen calibration
>   #define XPT2046_X_CALIBRATION 116903
>   #define XPT2046_Y_CALIBRATION 10346
>   #define XPT2046_X_OFFSET      -25
>   #define XPT2046_Y_OFFSET      -22
>   //#define TOUCH_ORIENTATION TOUCH_LANDSCAPE
> 
>   #if BOTH(TOUCH_SCREEN_CALIBRATION, EEPROM_SETTINGS)
>     #define TOUCH_CALIBRATION_AUTO_SAVE // Auto save successful calibration values to EEPROM
>   #endif
> 
>   #if ENABLED(TFT_COLOR_UI)
>     //#define SINGLE_TOUCH_NAVIGATION
>   #endif
> #endif
> ```

after a lot of working the issue, I eventually reinstalled over JTAG, and it started working.

TiNours38, 2023-01-25:
Hello
just a quick update. MKS Robin Nanon 1.2 + MKS TFT43 (FSMC). 
Touch screen calibration didn't work until I applied @rhapsodyv SPI patch : 
changed to 

> SPIx.Init.BaudRatePrescaler  = SPI_BAUDRATEPRESCALER_64;

And it works

screen & touch screen config : 

> #define MKS_ROBIN_TFT43
> ...
> #define TFT_COLOR_UI
> ...
> 
> define TOUCH_SCREEN
> #if ENABLED(TOUCH_SCREEN)
>   #define BUTTON_DELAY_EDIT      50 // (ms) Button repeat delay for edit screens
>   #define BUTTON_DELAY_MENU     250 // (ms) Button repeat delay for menus
> 
>   // #define DISABLE_ENCODER         // Disable the click encoder, if any
>   // #define TOUCH_IDLE_SLEEP_MINS 5 // (minutes) Display Sleep after a period of inactivity. Set with M255 S.
> 
>   #define TOUCH_SCREEN_CALIBRATION
> 
>   // // XPT2046 Touch Screen calibration
>   // M995 : 
>   // Recv: TOUCH_CALIBRATION_X 16291
>   // Recv: TOUCH_CALIBRATION_Y 11142
>   // Recv: TOUCH_OFFSET_X -9
>   // Recv: TOUCH_OFFSET_Y -33
> 
>   #define XPT2046_X_CALIBRATION 15817
>   #define XPT2046_Y_CALIBRATION 10748
>   #define XPT2046_X_OFFSET      -4
>   #define XPT2046_Y_OFFSET      -34
>   
>   #define TOUCH_ORIENTATION TOUCH_LANDSCAPE
> 
>   #if BOTH(TOUCH_SCREEN_CALIBRATION, EEPROM_SETTINGS)
>     #define TOUCH_CALIBRATION_AUTO_SAVE // Auto save successful calibration values to EEPROM
>   #endif
> 
>   #if ENABLED(TFT_COLOR_UI)
>     //#define SINGLE_TOUCH_NAVIGATION
>   #endif
> #endif

berkaybtr, 2023-05-17:
[Bjohnson131](https://github.com/Bjohnson131) How did you upload firmware over JTAG? I have tried every single thing but touch screen doesn't work, but I didn't try JTAG method.

Wickeda1985, 2024-09-21:
Hey ich have This Promlem  an have all the things do but it dont work can anybody upload The Firmware an Config ? Pls

thinkyhead, 2026-10-08:
Thanks for reporting this. A lot has changed in this area since it was filed. Could you please flash the current `bugfix-2.1.x` and tell us whether the problem still happens? If it does, please attach your current `Configuration.h` and `Configuration_adv.h` and a short description of the steps. We're trying to clear the decks for the 2.1.3 release, so a quick retest helps a lot.

closed 2026-10-20 by thinkyhead

HugReporter, 2026-11-10:
@rhapsodyv i'll name my kid after you