MarlinFirmware/Marlin #22210: [BUG] (Marlin 2.0.9: Heating instead of cooling in display)

alcomys, 2021-06-23:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Display show bed and nozzle cooling instead of heating with new Marlin 2.0.9
Version 2.0.8 does not have this issue.

### Bug Timeline

This is new for me

### Expected behavior

Expect to show heating instead of cooling.

### Actual behavior

Display show heating when cooling and vice verse 

### Version of Marlin Firmware

2.0.9

### Printer model

Anet A8 PLUS

### Electronics

SKR V1.4 

### Your Slicer

Cura

### Host Software

Pronterface

ellensp, 2021-06-24:
What lcd? What exactly are you talking about? Pictures showing the issue?
Please attach your configuration files.

alcomys, 2021-06-24:
Showing the issue?
if you read you know the problem,
my printer have only one display!
on that display it show when its heating and or cooling,
but when the heater is cooling it show heating and vice verse.
That's the problem

alcomys, 2021-06-24:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6708690/Marlin.zip)

closed 2021-06-24 by alcomys

ellensp, 2021-06-24:
Configs says its a standard REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and BOARD_BTT_SKR_V1_4
Which is good, as these are common components, so its easy to try and replicate.

I suspect your talking about the icon going solid or outline.
In which case its this settings that controls wich icon is which  #define STATUS_HOTEND_INVERTED      // Show solid nozzle bitmaps when heating (Requires STATUS_HOTEND_ANIM)

alcomys, 2021-06-24:
It shows the icon right, but only the text with the heating icon is not right. so when heating on the display say cooling but the temp is rizing, it looks like it is inverted