MarlinFirmware/Marlin #22212: [BUG] SKR v1.3 + MKS MINI12864 + onboard SD Creality CR20 (#15814)

creatron, 2021-06-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I upgraded my Creality CR20 Pro with silent steppers and SKR1.4 MB. Al works but I could either have the display or the uSD onboard to work, not both. I then found issue 15814 and followed the instructions to find that all work well if I include an additional line in the current bug fix version. 
In pins_BTT_SKR_V1_3.h
line ~380 
      #if ENABLED(MKS_MINI_12864)

        #define DOGLCD_CS            EXP1_05_PIN  // ok rs 1_05
        #define DOGLCD_A0            EXP1_04_PIN  // ok a0
        #define DOGLCD_SCK           EXP2_09_PIN  // sclk
        #define DOGLCD_MOSI          EXP2_05_PIN  // mosi

        //  if you have a Creality CR20  please add this line
        #define FORCE_SOFT_SPI    // >>>> <<<<<<<<

      #elif ENABLED(ENDER2_STOCKDISPLAY)

### Bug Timeline

Think is is an old issue, the above line was in the original 15814, but is not in the current code

### Expected behavior

Both the LCD display and uSD should work if enabled. 

### Actual behavior

The current version you can select LCD then only the LCD works, however if you select ONBOARD only the uSD works.  You now have to choose between onboard uSD or LCD display, they do not work at the same time.

### Version of Marlin Firmware

Creality CR20 Pro, upgraded with MB SKR1.3, using the stock mini 12864 clode LCD display

### Printer model

Creality CR20 Pro

### Electronics

Stock display, upgraded with SKR1.3 with TMC2208 (X,Y,Z) E0,E1 DRV8825, Neopixel, 

### Add-ons

NeoPixel..Problem only with CR20 mini clone 12864 display.. Ender stock, CR10 S4 etc works fine

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

creatron, 2021-08-01:
Nope, it still does not work, and I downloaded the zip file with the above
link, I only had to use my config and ini files.

Attach the zip file with config, ini and pins file with all the changes I
made. Basically, it is these lines in the
pins_BTT_SKR_V1_3.h file that is needed. line 387 for the above
      // >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
      # ifdef MKS_MINI_12864
      #  define FORCE_SOFT_SPI
      # endif

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.
Ask busy people, others are 'busy')

Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

creatron, 2021-10-01:
Was this now fixed?

creatron, 2021-12-02:
This bug is still present in the latest version of Marlin

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.
Ask busy people, others are 'busy')

Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

thinkyhead, 2022-04-21:
Maybe this configuration creates a conflict between multiple devices on the SPI chain, and perhaps one of them lacks a CS pin. Someone with a similar board / display will have to do some investigation.

creatron, 2022-04-24:
I have two of these setups, a CR20 pro with SKR MBs, I am an Engineer with embedded experience, and Segger emulator,  if someone can kick start me I am prepared to do the investigation.

closed 2022-07-04 by github-actions[bot]