MarlinFirmware/Marlin #22214: SWITCHING_TOOLHEAD_Z_HOP

madshammy, 2021-06-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Having an issue with getting SWITCHING_TOOLHEAD_Z_HOP to become active in my Marlin build.  I have #define MAGNETIC_SWITCHING_TOOLHEAD uncommitted.  upon build i keep getting an error,
[image]

### Bug Timeline

New

### Expected behavior

I expected this to follow the definition in line 308
#if ANY(SWITCHING_TOOLHEAD, MAGNETIC_SWITCHING_TOOLHEAD, ELECTROMAGNETIC_SWITCHING_TOOLHEAD)

### Actual behavior

[image]

### Steps to Reproduce

#define MAGNETIC_SWITCHING_TOOLHEAD
had to bring this line up to make work

[image]

### Version of Marlin Firmware

#define CONFIGURATION_H_VERSION 02000900

### Printer model

Anycubic

### Your Slicer

Cura

yopla31, 2021-06-26:
I opened an issu ([[BUG] Avec MAGNETIC_SWITCHING_TOOLHEAD, erreur de compilation #19270](https://github.com/MarlinFirmware/Marlin/issues/19270#issue-694144845)) on this and my solution was the same (adding the missing line). My remark was not considered and the problem closed.
With each release, I add this missing line !

ellensp, 2021-06-26:
This is not the correct fix.

SWITCHING_TOOLHEAD_Z_HOP is not required by the MAGNETIC_SWITCHING_TOOLHEAD code only in the ELECTROMAGNETIC_SWITCHING_TOOLHEAD code.

The sanity test is incorrect

```DIFF
+++ b/Marlin/src/inc/SanityCheck.h
@@ -1256,15 +1256,15 @@ static_assert(Y_MAX_LENGTH >= Y_BED_SIZE, "Movement bounds (Y_MIN_POS, Y_MAX_POS
     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Y_POS"
   #elif !defined(SWITCHING_TOOLHEAD_X_POS)
     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_X_POS"
-  #elif !defined(SWITCHING_TOOLHEAD_Z_HOP)
-    #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
   #elif !defined(SWITCHING_TOOLHEAD_Y_CLEAR)
     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Y_CLEAR."
   #elif ENABLED(ELECTROMAGNETIC_SWITCHING_TOOLHEAD)
     #if ENABLED(EXT_SOLENOID)
-      #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD and EXT_SOLENOID are incompatible. (Pins are used twice.)"
+      #error "ELECTROMAGNETIC_SWITCHING_TOOLHEAD and EXT_SOLENOID are incompatible. (Pins are used twice.)"
     #elif !PIN_EXISTS(SOL0)
-      #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SOL0_PIN."
+      #error "ELECTROMAGNETIC_SWITCHING_TOOLHEAD requires SOL0_PIN."
+    #elif !defined(SWITCHING_TOOLHEAD_Z_HOP)
+      #error "ELECTROMAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
     #endif
   #endif
 #endif

```

@yopla31 please try and remember Marlin is maintained by volunteers! Giving up their free time to help.
Try to have patience and understanding.

madshammy, 2021-06-26:
Thanks for the replay @ellensp I am still pretty new to this stuff.  I am unsure how to bypass the sanity checks I'll try to read up more on it.  I am sure with the PrinterMods Xchange system there will be a few people that will be running into this problem soon enough so glad we are getting it addressed.  Thanks to all you folks that take the time to hammer out these problems it's much appreciated.

ellensp, 2021-06-26:
@madshammy try and read the DIFF file above
you edit Marlin/src/inc/SanityCheck.h
delete any re lines, add in the green lines.
Don't include the +'s or -'s  at the start of the red or green lines.

I've also created a PR, so this will be fixed in marlin shorty.

madshammy, 2021-06-26:
Thanks very much for the help, was able to do those changes and worked like a charm.  Do I close this this out or is that done once the changes are take effect?

[image]

ellensp, 2021-06-26:
Leave open till changes are in Marlin.

yopla31, 2021-06-26:
> @yopla31 please try and remember Marlin is maintained by volunteers! Giving up their free time to help.
> Try to have patience and understanding.

@ellensp , it is very important to remind each of us that the community is working on a voluntary basis for the evolution of Marlin. It is a collegial work where the users report possible problems, propose evolutions, propose workarounds to the problems, and others fix the code.
And it is in this state of mind that I did not insist more when my request was closed.
Your fix is much more elegant than the workaround proposed by madshammy and myself. And I would have preferred to have this answer to my request.
Respect and thanks for your work!

ellensp, 2021-06-27:
This has now been merged into marlin. Closing.

closed 2021-06-27 by ellensp