MarlinFirmware/Marlin #22216: [BUG] STM32 - error: 'setInterruptPriority' is not a member of 'SoftwareSerial'

dotdash32, 2021-06-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Tried compiling in both 2.0.9 and bugfix-2.0.x, but ran into issues.  It looks like its in the STM32 HAL, my board is an SKR E3 DIP.  I am now getting this error:

```
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/STM32/timers.cpp:24:
Marlin/src/HAL/STM32/timers.cpp: In function 'void SetTimerInterruptPriorities()':
Marlin/src/HAL/STM32/timers.cpp:190:44: error: 'setInterruptPriority' is not a member of 'SoftwareSerial'
  190 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |                                            ^~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:558:26: note: in definition of macro 'THIRD'
  558 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin/src/HAL/STM32/../../inc/../core/macros.h:204:29: note: in expansion of macro '___TERN'
  204 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:203:29: note: in expansion of macro '__TERN'
  203 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:202:29: note: in expansion of macro '_TERN'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin/src/HAL/STM32/timers.cpp:190:3: note: in expansion of macro 'TERN_'
  190 |   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
      |   ^~~~~
*** [.pio/build/STM32F103RC_btt_USB/src/src/HAL/STM32/timers.cpp.o] Error 1
```

Seems somewhat similar to #17823, but I think it's a different cause, similar symptom?

I also checked my branch for an SKR 1.4T, and that compiles fine.

### Bug Timeline

Was able to compile 2.0.8 ish?  

### Expected behavior

Compilation without errors

### Actual behavior

Errors during compliation, see above

### Steps to Reproduce

1. Set board as SKR E3 DIP
2. Compile (Using ABM, any STM_BTT version)
3. oof owie my errors

### Version of Marlin Firmware

2.0.9

### Printer model

Ender 3

### Electronics

SKR E3 DIP

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6706058/Archive.zip)

[Also see the forked branch](https://github.com/dotdash32/Marlin/tree/Ender3_skrE3DIP_bugfix)

ellensp, 2021-06-24:
I cannot reproduce this using provided configs on bugfix
Delete your .pio directory and try again.

ellensp, 2021-06-24:
Forked branch also compiles without issues,

closed 2021-06-24 by ellensp

dotdash32, 2021-06-24:
Leaving solution as a reference: deleting .pio directory and re-building worked.  Thanks.