MarlinFirmware/Marlin #22217: MINTEMP error when setting temp from LCD panel with high-temp thermistor installed

p-v-n, 2021-06-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With high-temp thermistor and hotend completely cold (thermistor is not warm enough to provide a good reading), an attempt to set temp from LCD panel on the printer always results in MINTEMP error and printer halting.
Note: this does NOT happen when setting the temp using M109 command.

The following defines are set in Configuration_adv.h:

```
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10
#define MILLISECONDS_PREHEAT_TIME 30000
```
(full configuration files are attached)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6707803/Marlin.zip)

### Bug Timeline

Started as soon as I installed a new high-temp thermistor, so potentially an issue that existed long before that

### Expected behavior

Hotend temperature is set and it starts heating

### Actual behavior

Printer halts with "MINTEMP" error

### Steps to Reproduce

1. Install high-temp thermistor
2. Build Marlin using attached config
3. Make sure hotend is completely cold
4. From printer's LCD panel try to set some temperature, say 200°C.

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Jun 23 2021 01:28:22)

### Printer model

Geeetech A10M

### Electronics

All stock

### Add-ons

High-temp thermistor

slowbro, 2021-06-26:
When heating, does the displayed temperature change? What temp does it get to/how long before it errors?

p-v-n, 2021-06-26:
@slowbro , when using GCODE command, everything is normal - as soon as hotend is warm enough for the thermistor to provide good reading (starts around 30°C), the temp is displayed consistently.
However, when setting the temp from LCD panel, printer halts immediately.

slowbro, 2021-06-27:
Alright, thanks for the follow-up. I'll see what I can find.

slowbro, 2021-06-27:
@p-v-n This patch should fix that issue, can you try it out?

```diff
diff --git a/Marlin/src/lcd/menu/menu_temperature.cpp b/Marlin/src/lcd/menu/menu_temperature.cpp
index 96e6ee5f8e..768d894e4f 100644
--- a/Marlin/src/lcd/menu/menu_temperature.cpp
+++ b/Marlin/src/lcd/menu/menu_temperature.cpp
@@ -169,10 +169,10 @@ void menu_temperature() {
   // Nozzle [1-5]:
   //
   #if HOTENDS == 1
-    EDIT_ITEM_FAST(int3, MSG_NOZZLE, &thermalManager.temp_hotend[0].target, 0, thermalManager.hotend_max_target(0), []{ thermalManager.start_watching_hotend(0); });
+    EDIT_ITEM_FAST(int3, MSG_NOZZLE, &thermalManager.temp_hotend[0].target, 0, thermalManager.hotend_max_target(0), []{ thermalManager.setTargetHotend(thermalManager.temp_hotend[0].target, 0); });
   #elif HAS_MULTI_HOTEND
     HOTEND_LOOP()
-      EDIT_ITEM_FAST_N(int3, e, MSG_NOZZLE_N, &thermalManager.temp_hotend[e].target, 0, thermalManager.hotend_max_target(e), []{ thermalManager.start_watching_hotend(MenuItemBase::itemIndex); });
+      EDIT_ITEM_FAST_N(int3, e, MSG_NOZZLE_N, &thermalManager.temp_hotend[e].target, 0, thermalManager.hotend_max_target(e), []{ thermalManager.setTargetHotend(thermalManager.temp_hotend[MenuItemBase::itemIndex].target, MenuItemBase::itemIndex); });
   #endif
 
   #if ENABLED(SINGLENOZZLE_STANDBY_TEMP)
```

I'm not sure that's the best solution permanently, but the issue seems to be the LCD temp-settings skipping the calls to `reset_preheat_time()` and `start_preheat_time()` in `setTargetHotend()`:

https://github.com/MarlinFirmware/Marlin/blob/81ca2dd273358a0cfc0f65d5b4900d83abc96d69/Marlin/src/module/temperature.h#L661-L672

CRCinAU, 2021-06-28:
I'm not sure what SDK this setup uses, but there was an issue with optimisation that was fixed recently with the STM32 libraries that would cause this same error.

Make sure you're building from the latest bugfix-2.0.x and remove your temp directories (like ~/.platformio/) and confirm if this still occurs.

slowbro, 2021-06-28:
@CRCinAU it's an AVR.. mega2560 I believe.

@p-v-n ok, I tested this change on my SKR with basically the same options (though, I swapped out SINGLENOZZLE for 2 extruders so I could test the HOTEND_LOOP...); it seems to work, and I don't get an immediate MINTEMP error:

```diff
diff --git a/Marlin/src/lcd/menu/menu.h b/Marlin/src/lcd/menu/menu.h
index 290832c49c..28d377da0c 100644
--- a/Marlin/src/lcd/menu/menu.h
+++ b/Marlin/src/lcd/menu/menu.h
@@ -132,14 +132,15 @@ class MenuItem_confirm : public MenuItemBase {
 
 // The Menu Edit shadow value
 typedef union {
-  bool     state;
-  float    decimal;
-  int8_t   int8;
-  int16_t  int16;
-  int32_t  int32;
-  uint8_t  uint8;
-  uint16_t uint16;
-  uint32_t uint32;
+  bool      state;
+  float     decimal;
+  int8_t    int8;
+  int16_t   int16;
+  int32_t   int32;
+  uint8_t   uint8;
+  uint16_t  uint16;
+  uint32_t  uint32;
+  celsius_t celsius;
 } chimera_t;
 extern chimera_t editable;
 
diff --git a/Marlin/src/lcd/menu/menu_temperature.cpp b/Marlin/src/lcd/menu/menu_temperature.cpp
index 96e6ee5f8e..65cef5b76d 100644
--- a/Marlin/src/lcd/menu/menu_temperature.cpp
+++ b/Marlin/src/lcd/menu/menu_temperature.cpp
@@ -169,10 +169,13 @@ void menu_temperature() {
   // Nozzle [1-5]:
   //
   #if HOTENDS == 1
-    EDIT_ITEM_FAST(int3, MSG_NOZZLE, &thermalManager.temp_hotend[0].target, 0, thermalManager.hotend_max_target(0), []{ thermalManager.start_watching_hotend(0); });
+    editable.celsius = thermalManager.temp_hotend[0].target;
+    EDIT_ITEM_FAST(int3, MSG_NOZZLE, &editable.celsius, 0, thermalManager.hotend_max_target(0), []{ thermalManager.setTargetHotend(editable.celsius, 0); });
   #elif HAS_MULTI_HOTEND
-    HOTEND_LOOP()
-      EDIT_ITEM_FAST_N(int3, e, MSG_NOZZLE_N, &thermalManager.temp_hotend[e].target, 0, thermalManager.hotend_max_target(e), []{ thermalManager.start_watching_hotend(MenuItemBase::itemIndex); });
+    HOTEND_LOOP() {
+      editable.celsius = thermalManager.temp_hotend[e].target;
+      EDIT_ITEM_FAST_N(int3, e, MSG_NOZZLE_N, &editable.celsius, 0, thermalManager.hotend_max_target(e), []{ thermalManager.setTargetHotend(editable.celsius, MenuItemBase::itemIndex); });
+    }
   #endif
 
   #if ENABLED(SINGLENOZZLE_STANDBY_TEMP)
```

Let me know if that works for you.

p-v-n, 2021-06-28:
@slowbro, this works!

slowbro, 2021-06-28:
Great! I'll submit a PR.

PR #22247 (2021-06-28): Use setTargetHotend in temperature menu
### Description

When setting a temperature in the menus, the preheat timer is bypassed due to setting the target directly and immediately calling `start_watching_hotend(n)`. Instead, use the `editable` field to get the desired temperature and use `Temperature::setTargetHotend()` so that related calls happen.

Not sure if adding `celsius_t` to `chimera_t` is wanted, since the base unit is `int16_t`, but I think it makes it read better (and doesn't require one to know that connection between celsius and int16).

### Requirements

LCD, high resistance temp [... cut]
closed 2021-06-28 by thinkyhead