MarlinFirmware/Marlin #22218: [BUG] u8g.h error

Enzo-Enz, 2021-06-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi guys
after fighting with arduino ide and install u8glib hal master,i faced with another issue:

Arduino: 1.8.15 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\USER\Downloads\Marlin-2.0.x\Marlin\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp:65:0:

C:\Users\USER\Documents\Arduino\libraries\U8glib-HAL-master/U8glib.h:47:10: fatal error: clib/u8g.h: No such file or directory

 #include "clib/u8g.h"

          ^~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

is there any way to solve this error?
unfortunately because of some reason I can't use visual stadio

### Version of Marlin Firmware

2.0.9

### Your Slicer

Cura

ellensp, 2021-06-24:
Seriously use platformio under vscode. It is 10 times faster to compile Marlim than the primitive Arduino IDE 
https://marlinfw.org/docs/basics/install_platformio_vscode.html

But if you absolutely insist on using Arduino IDE 

Firstly uninstall standard U8glib from the Arduino IDE library manager.
Download the latest U8glib-HAL https://github.com/MarlinFirmware/U8glib-HAL/archive/refs/heads/master.zip 
Extract the directory into your Arduino/libraries folder
Download [library.properties.zip](https://github.com/MarlinFirmware/Marlin/files/6710738/library.properties.zip)
Extract library.properties to Arduino/libraries/U8glib-HAl-master 
[Restart Arduino IDE so it notices libraries have changed. 
Click upload and take a long nap while arduino ide slowly compiles.

closed 2021-06-24 by ellensp

Enzo-Enz, 2021-06-25:
@ellensp Thank you so much bro,It worked perfect and I updated my printer

ellensp, 2021-07-11:
Update. 

Marlin has been updated to need U8glib-HAL
This removes the conflict with the old no longer compatible U8glib
Just search for U8glib-HAL in the library manager and install it, That is all that is needed.

No more having to manually download files and add in a library.properties files etc.