MarlinFirmware/Marlin #22220: G38 will not stop probe moving regardless of if its triggered

xxandervargad, 2021-06-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

We are trying to make a CNC machine that stops when it comes in contact with an object. There is no z_probe input on the board that we're using, so we have `Z_MIN_PROBE_USES_Z_ENDSTOP_PIN` enabled. 

The problem is that G38 moves (both G38.2 and G38.3) do not stop when the Z endstop is triggered. 
It also has started the move despite the endstop already being triggered

I tried using M120 to enable endstops before trying G38. I have also confirmed that they work correctly with M119

Thanks for your help

### Bug Timeline

Has been a problem for a few months

### Expected behavior

Probe stops when Z_min in triggered

### Actual behavior

Probe does not stop until it reaches destination, basically identical behavior to G0 or G1

### Steps to Reproduce

1. Use a [Rambo Board](https://ultimachine.com/products/rambo-1-4) to make a CNC
2. enable `Z_MIN_PROBE_USES_Z_ENDSTOP_PIN`
3. send command "G38 F500 X200" as a test
4. trigger Z_min manually
5. watch sadly as the probe ignores you and continues to move right 

### Version of Marlin Firmware

2.0.7. It is not up to date because 2.0.9 wont compile for some unrelated reason

### Printer model

Homeade CNC

### Electronics

Rambo Board

### Host Software

Other (explain below)

### Additional information & file uploads

[here are the header files](https://github.com/MarlinFirmware/Marlin/files/6711876/G38_will_not_stop_probe_moving_regardless_of_if_its_triggered.zip)

Prophet-of-Doom, 2021-06-24:
I'm experiencing the same issue on both accounts. My 2.09 won't compile for my rambo too. I've delved into this before months ago and did everything short of opening a github issue but ultimately went a different direction. Hope someone can provide a proper solution.

thisiskeithb, 2021-06-26:
"Trigger existing endstops on G38 hit" (https://github.com/MarlinFirmware/Marlin/commit/500b7872fb92d35dd0d1a690f4e5cf1aadea2f46) likely resolves this, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

xxandervargad, 2021-07-02:
We have uploaded `bugfix-2.0.x` and it did not seem to fix anything. 

I've attached the new configuration files and a write-up of our procedure to reproduce.
```
G38.2 command executes even while Z_min is still triggered after homing.
prints "Error: Failed to reach target"

Robot is moved so that no endstops are triggered. (G0 X5 Y5 Z5)
G38.2 X100 F500 is executed. Half way through, Z_min endstop is triggered.
gantry continues to move to destination and prints "Error: Failed to reach target"

G38.2 X0 F500 is then excecuted WHILE Z_min is manually triggered.
Half way, Z_min is un-triggered. Gantry continues to X0 and prints "Error: Failed to reach target"
```

[Files.zip](https://github.com/MarlinFirmware/Marlin/files/6756050/Files.zip)

xxandervargad, 2021-07-14:
Still waiting for a reply.

xxandervargad, 2021-07-19:
@thisiskeithb just checking if you could still help us with this issue. It is currently one of our development bottlenecks.

DerAndere1, 2021-07-27:
Try the following changes and report if it works:
https://github.com/MarlinFirmware/Marlin/blob/fc4f4ab5f5e47b5ac764ca6c2fe127dc9ecbf55d/Marlin/src/module/probe.cpp#L518
change to:
```
      TEST(endstops.trigger_state(), TERN(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, Z_MIN, Z_MIN_PROBE))
```

and
https://github.com/MarlinFirmware/Marlin/blob/fc4f4ab5f5e47b5ac764ca6c2fe127dc9ecbf55d/Marlin/src/module/endstops.h#L72
change to: 
``` 
  _ES_ITEM(HAS_Z_MIN_PROBE_PIN, Z_MIN_PROBE)

```

Prophet-of-Doom, 2021-07-28:
@DerAndere1 
Thank you for the suggestion, I have good news and bad news! 
Bad news: your suggestion didn't work, mainly because it wouldn't compile with that second change. I modified it till I got it to compile but I experienced the same issue. 

Good news: I did get it to work. We are using version 2.6.1 for unrelated reasons regarding mixing stepper where they don't work in the recent bugfix version. I changed the Configuration.h (same as the one originally posted) to disable 
`//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
and enable and change the pin number of
`#define Z_MIN_PROBE_PIN 30 // to match zmax of RAMBO 1.4, was 32 RAMPS default`
I suppose I could also define it to use the Z_MIN pin number but I haven't tested that. Additionally I have `#define NOZZLE_AS_PROBE`

From my understanding, G38 should ideally work with just defining Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN as it seems to be the only feature that doesn't seem to recognize it. 

As for the most recent bugfix-2.0.x, I delved into it a bit more thoroughly:
Using the same configuration changes as above, I traced it to these same lines. 
https://github.com/MarlinFirmware/Marlin/blob/296d0d495cad942cccd611b74a6b8a414a089a65/Marlin/src/module/endstops.cpp#L618 https://github.com/MarlinFirmware/Marlin/blob/296d0d495cad942cccd611b74a6b8a414a089a65/Marlin/src/module/endstops.cpp#L933 Here, `HAS_Z_MIN_PROBE_PIN` is 0. This is due to:
https://github.com/MarlinFirmware/Marlin/blob/296d0d495cad942cccd611b74a6b8a414a089a65/Marlin/src/inc/Conditionals_post.h#L2169-L2171 Wherein `HAS_BED_PROBE` is 1, `HAS_CUSTOM_PROBE_PIN` is 0, and `PIN_EXISTS(Z_MIN_PROBE)` is 0.

The logic behind this whole check isn't really obvious to me. The check for `HAS_Z_MIN_PROBE_PIN` shouldn't rely on `HAS_CUSTOM_PROBE_PIN`, as the existence of a custom probe pin relies on `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` being disabled. https://github.com/MarlinFirmware/Marlin/blob/d5a9a04abecf6eaf819e9ccabd6702dccd984a9a/Marlin/src/inc/Conditionals_LCD.h#L913-L915 Additionally, `Z_MIN_PROBE` is being set to -1 as the check to disable unused pins doesn't take into account `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` https://github.com/MarlinFirmware/Marlin/blob/d5a9a04abecf6eaf819e9ccabd6702dccd984a9a/Marlin/src/pins/pins_postprocess.h#L1137-L1140  

As for a fix, I changed
https://github.com/MarlinFirmware/Marlin/blob/296d0d495cad942cccd611b74a6b8a414a089a65/Marlin/src/module/endstops.cpp#L618 to:
`#if ENABLED(G38_PROBE_TARGET) && EITHER(HAS_Z_MIN_PROBE_PIN, HAS_CUSTOM_PROBE_PIN) && NONE(CORE_IS_XY, CORE_IS_XZ, MARKFORGED_XY)`
then this:
https://github.com/MarlinFirmware/Marlin/blob/d5a9a04abecf6eaf819e9ccabd6702dccd984a9a/Marlin/src/inc/Conditionals_LCD.h#L913-L915 to: 
`#if ENABLED(HAS_BED_PROBE) && PIN_EXISTS(Z_MIN_PROBE)`
and finally removed this block entirely as not only is it nonsensical to disable a custom pin if it isn't defined, but also to arbitrarily assume the custom pin would be using `Z_MIN` 
https://github.com/MarlinFirmware/Marlin/blob/d5a9a04abecf6eaf819e9ccabd6702dccd984a9a/Marlin/src/pins/pins_postprocess.h#L1137-L1140 

@thisiskeithb I'm going to submit this as a pull request (my first one!) if you'd like to confirm my findings. 

I should clarify, @xxandervargad and I are both working on the same project. Don't know why I posted like I didn't, must've been a long day! Also for better visualization as to what we're making, it's essentially a router that works on pre-cut intricate blanks with crevices that the tool head needs to get into without damaging. We have a sensor that triggers an endstop when the tool head bumps into an object.

thinkyhead, 2021-07-29:
I have modified the PR to see if we can simply use the designated probe pin, because the original design of `G38` always uses the same probe as `G29`, `G30`, etc.

closed 2021-07-30 by thinkyhead