MarlinFirmware/Marlin #22225: [BUG] (Flow Sensor Compile Error)

vr6syncro, 2021-06-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If i enable the #define LASER_COOLANT_FLOW_METER firmware compile throw a error.

`Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::manage_heater()':
Marlin\src\module\temperature.cpp:1554:12: error: 'class MarlinUI' has no member named 'flow_fault'
 1554 |         ui.flow_fault();
      |            ^~~~~~~~~~
Compiling .pio\build\LPC1769\src\src\sd\SdFile.cpp.o
Compiling .pio\build\LPC1769\src\src\sd\SdVolume.cpp.o
*** [.pio\build\LPC1769\src\src\module\temperature.cpp.o] Error 1`

### Bug Timeline

new

### Expected behavior

compile without errors

### Actual behavior

compile with errors

### Steps to Reproduce

just enable the #define LASER_COOLANT_FLOW_METER

### Version of Marlin Firmware

latest github

### Printer model

-

### Electronics

-

### Add-ons

-

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6719959/Marlin.zip)
Attached the config files.

I did not use a LCD if the error MarlinUI means a feature or so for that.

ellensp, 2021-06-26:
The code presumes the existence of a LCD
This allows it to compile. 

```DIFF
diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
index 9f32ce933b..409fd72a00 100644
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -1551,7 +1551,9 @@ void Temperature::manage_heater() {
     #if ENABLED(FLOWMETER_SAFETY)
       if (cutter.enabled() && cooler.check_flow_too_low()) {
         cutter.disable();
-        ui.flow_fault();
+        #if HAS_DISPLAY
+          ui.flow_fault();
+        #endif
       }
     #endif
   #endif

```

ellensp, 2021-06-26:
also there is no error sent to serial on check_flow_too_low failing, it  silently disables the cutter.

ellensp, 2021-06-26:
@descipher can you update so lcd is not requirement?

descipher, 2021-06-26:
This code does indeed need a display and that was the intent. It supports 12864 and 2004 basic displays. How do you see the flow rate without one? We could send status data via serial. It does already send a fault message via serial and will not stop until the fault is fixed.
The error message is sent when the user sends any M3 based power.

Is this use case only to detect a flow fault for tube safety without a display?

vr6syncro, 2021-06-26:
Yes, the goal was only to detect a flow fault and stop the laser. 
Background Info: Display is a nice to have. But in my mind i want to check it with a webinterface and a esp chip. this is already working and should display a faulty flow in the console. but it is to early  in my development to say if it is the best practise or working well because i have not testet it in real. i get the firmware done first.

descipher, 2021-06-26:
Ok if I understand correctly it sounds like your use case is to control the laser using a wifi connected web ui like the ESP3D and you also want to monitor flow meter data or state.

For the short term we can do a PR to enable the code the run without a display. Longer term should have some method to allow a pull or push data via serial in some already established  Marlin format. 
So currently with this change you would only receive a serial text error as an indication of a flow fault when a power level more than 0 is sent via gcodes. 

Are you good with this in the short term?

vr6syncro, 2021-06-26:
Yes that would be good. Thanks for the quick fix.

closed 2021-07-30 by thinkyhead