MarlinFirmware/Marlin #22226: [FR] Provide binary releases for common stock configurations

USA-RedDragon, 2021-06-26:
### Is your feature request related to a problem? Please describe.

With 3d printers rapidly becoming more and more affordable, the bar for entry past stock firmware is very, very high, requiring a lot of learning in how to configure Marlin, and/or lots of frustration for users who aren't quite programming-savvy, releasing prebuilt "drop-in" binaries for common 3D printer configurations (such as the Ender 3 line) based on the configurations already maintained within https://github.com/MarlinFirmware/Configurations.

### Are you looking for hardware support?

N/A

### Describe the feature you want

Providing "drop-in" binaries for common off-the-shelf 3d printers (the Ender 3 line is a great example) would lower the bar for entry quite low compared to any other method possible. This could also provide the benefit of being able to leverage CI to ensure that the configurations published in https://github.com/MarlinFirmware/Configurations continue to be tested against the main repository, ensuring that they don't desynchronize or have build errors by default. 

### Additional context

If this is something desirable, I can gladly introduce CI (GitHub Actions) scripts as a PR, leveraging GitHub Releases to distribute the binaries. I have a CI workflow on my personal fork of Marlin to auto build any changes I make and release a binary for me to easily push to the printer. https://github.com/USA-RedDragon/Marlin/blob/main/.github/workflows/autobuild.yml

CRCinAU, 2021-06-27:
That was my idea behind: https://marlin.crc.id.au/

To be clear, it isn't associated with the Marlin FW specifically, I do use some of the subscription cost to sponsor developers who work on Marlin.

It has caught numerous bugs in the past as well - which has the added benefit of getting newer code testing as well and get issues identified and resolved faster.

I'm looking to expand it into different printer manufacturers - however without physical hardware or testers, this is a little more difficult.

USA-RedDragon, 2021-06-30:
That's awesome @CRCinAU, basically exactly what I envisioned when making this FR. 

Evidently, the benefits of a system like this are great, and with upstream support of something like this, the community would be able to help in expansion to different manufacturers.

thinkyhead, 2021-07-30:
It is definitely preferred to make this something external to the Marlin project and we encourage anyone who wants to do this to carry on. We are already maintaining a huuuuge pile of Configurations, and this suggestion is only going to multiply those, and the maintenance problem for a tiny core of maintainers with limited time and resources.

USA-RedDragon, 2021-07-30:
The idea shouldn't conflict with maintenance of configuration files, rather it compliments the process. The request isn't to maintain new configuration files, rather utilize the many, many configurations that are already maintained that work for stock boards. The idea is to use continuous integration to ensure that these stock configurations do at least compile (more rich continuous integration, such as ensuring the boards boot are possible, but would obviously require a sizable amount of work. This could also be utilized to test the impact of potentially breaking changes, and as a sanity check for pull requests, leading to less maintenance tasks required when accepting community changes

thinkyhead, 2021-07-30:
I worry that it would encourage fans of pre-built binaries to start adding more configs with a small handful of tweaks to support BLTouch versus no-BLTouch, and every other combination of hardware, until we have an even vaster pile of configurations. And I expect it will generate a flood of adjustments from many people who don't think the binary is quite right. Then there is the question of how are these binaries going to be built and uploaded, and who will do that?

It might be fairly trivial to make the scripts to do all the builds, but someone has to run them someplace and re-run them as configurations change and are fixed, and so on. I am very much already personally overwhelmed with the workload this project presents, and not anxious to take on even more maintenance tasks, so someone else must take point if there really is a strong desire for pre-built binaries that users will prefer over ones that they have customized.

I will leave this FR open, but truly I cannot take it on myself.

thinkyhead, 2025-10-02:
What can I say? I finally warmed to the idea once it crossed a certain threshold of easy.