MarlinFirmware/Marlin #22233: SCARA mode - X and Y won't home

DejanN7, 2021-06-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm not sure whether it's a bug or it's me not knowing how to set up the SCARA configuration, but the X and Y axis won't go to home position, while Z homing works fine.
I'm using simple NC microswitches, with "#define USE_XMIN_PLUG", Y and Z, set to MIN, and "#define X_HOME_DIR -1", Y and Z set the -1. For example, when trying the default Configuration.h for a normal cartesian printer, and click the home button on Pronterface, the homing works for all three axis. But when I try it with SCARA configuration it doesn't, I get the following error:
echo:Homing Failed
Error:Printer halted. kill() called!
Error:Printer halted. kill() called!
If I comment "#define VALIDATE_HOMING_ENDSTOPS", I don't get the error, but X and Y won't home while Z will home properly.
So any idea what could cause the problem and where should I look to make adjustments to the code?

### Version of Marlin Firmware

2.0.9

### Electronics

RAMPs 1.6, Arduino MEGA

### Host Software

Pronterface

GMagician, 2021-06-26:
Check with M119 it seems like you have reversed signals (trigger means you are at home)

ellensp, 2021-06-27:
Have you done basic diagnosis ie send the machine a M119 to check endstop status?

DejanN7, 2021-06-28:
The endstops status is fine, open, and when I press them they are triggered. Just like I said when trying the default configuration files, the homing works for all axis, they home properly which means the endstops work fine. But when using the SCARA configuration files it doesn't work.
Here are my configuration files:
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/6725209/Configuration.files.zip)

Another note is that after a week of trying to get it working, I realized that there might be an error in the scara.cpp. My SCARA is a single arm, not parallel, which means I should be using the MP_SCARA. However I couldn't get the kinematics right. But when I used the MORGAN_SCARA kinematics (by replacing line 73 in scara.cpp, from #if ENABLED(MORGAN_SCARA) to MP_SCARA) the kinematics worked properly. So actually I'm using the MORGAN_SCARA inverse kinematics which is a parallel SCARA, for my single arm SCARA. 
Maybe the MP_SCARA inverse kinematics work properly but I don't know how to set it up.
For setting up the MORGAN_SCARA inverse kinematics I'm using the following adjustments for defining the home position of the arm, from where the inverse kinematics is then calculated ( homing is still not working, I just manually set the arm to position):
The arms lengths:
#define SCARA_LINKAGE_1  228    // (mm)
#define SCARA_LINKAGE_2 136.5    // (mm)
The bed zero position relative to the shoulder joint:
#define SCARA_OFFSET_X    50       // (mm)
#define SCARA_OFFSET_Y    -150    // (mm)
and the nozzle location relative to the bed zero position:
#define MANUAL_X_HOME_POS 247.45
#define MANUAL_Y_HOME_POS -127.5
With these settings I managed to get the kinematics working properly, but just like I said, by actually using the MORGAR_SCARA inverse kinematics in the scara.cpp instead of MP_SCARA.

svsergo, 2021-06-28:
Look  [BUG] MP_SCARA doesn't work #20764

ellensp, 2021-06-28:
@svsergo that was fied https://github.com/MarlinFirmware/Marlin/pull/21113
This new issue is probably related to the new 6 axis marlin

DejanN7, 2021-06-28:
@svsergo Yes, I looked at #20764 which actually helped me to realize how to set it up. With the latest version the situation is just like I explained above.
What about the homing, how does it work for you?

svsergo, 2021-06-28:
For MP_SCARA homing, a standard function is used. This also works in the latest version. Try it in the Configuration settings.h
close the line //#define VALIDATE_HOMING_ENDSTOPS

DejanN7, 2021-06-29:
I tried that but it doesn't work. X and Y don't home, only Z axis do in such a case.

DejanN7, 2021-06-29:
I found some kind of solution for the homing.
In motion.cpp I commented line 1716 and 1717:
 1716.  //if (axis != Z_AXIS) { BUZZ(100, 880); return; }
 1717. //#else
Also in Configuration.h //#define VALIDATE_HOMING_ENDSTOPS should be commented.
In this way, the homing for my SCARA is working.

DerAndere1, 2021-07-24:
This seems to no bug but be expected behaviour according to the documentation in the source code:
https://github.com/MarlinFirmware/Marlin/blob/b925130db13409ba1ee671dea151c81990668d8b/Marlin/src/module/motion.cpp#L1724-L1725

thinkyhead, 2021-07-30:
SCARA is generally not fully baked due to very limited interest. I did a load of work for a SCARA that never made it to production, and in that course made several adjustments to homing and other things. Possibly at some point I will be able to port those over. In general, it was simplest to home Z to MAX as the first step in SCARA homing, and then the arm(s) would be most safe to home afterward. The homing of A and B were specified with angles, and the kinematics to convert between angles and positions was fairly complete, though it did not use the typical SCARA trigonometry. If there is great interest, get in touch with me on Discord and I will pass on any notes I have about SCARA to help you get it closer to working.

thinkyhead, 2021-07-30:
P.S., There might be generally better and less broken SCARA support in Marlin 1.1.9 so you might want to try that and see how close it is.