MarlinFirmware/Marlin #22235: [BUG] (PSU_Control without Hotend)

vr6syncro, 2021-06-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Enabled #define PSU_CONTROL
#define AUTO_POWER_CONTROL
#define POWER_TIMEOUT  

Only using #define TEMP_SENSOR_COOLER 1

Have no Extruder or Hotend because i am using Laser, so i disabled it.

When i enabled auto_power_control the code copliles with error:
`Marlin\src\feature\power.cpp: In static member function 'static bool Power::is_power_needed()':
Marlin\src\feature\power.cpp:84:77: error: 'class Temperature' has no member named 'temp_hotend'
   84 |   HOTEND_LOOP() if (thermalManager.degTargetHotend(e) > 0 || thermalManager.temp_hotend[e].soft_pwm_amount > 0) return true;
      |                                                                             ^~~~~~~~~~~`
With auto_power_control disabled it compiles fine.

### Bug Timeline

new

### Expected behavior

compile with no errors.
(After 30 seconds M81 in idle.)

### Actual behavior

compile with error

### Steps to Reproduce

#define PSU_CONTROL
#define AUTO_POWER_CONTROL

disable extruder and any hotend stuff

### Version of Marlin Firmware

latest github

### Printer model

-

### Electronics

-

### Add-ons

-

### Additional information & file uploads

I am unsure about the function PSU_CONTROL maybe i missed something in the code?
Attached my config.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6721323/Marlin.zip)

slowbro, 2021-06-27:
Found the issue, will submit a PR.

slowbro, 2021-06-28:
#22245 was merged, this should be resolved.

closed 2021-06-28 by thisiskeithb