MarlinFirmware/Marlin #22236: [BUG] Unified Bed Leveling not probing full bed area

meltonpieman, 2021-06-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have an Ender 3 pro with a bed width of 235mm. However, the print head can move to X260 before mechanical interference. UBL with the MESH_INSET 20 gives a very limited probing (10x10 gets anly 6x6). 

### Bug Timeline

In 2.0.7 when I first used UBL however 2.0.9 seemed to offer more ...

### Expected behavior

I expected to be able to use the BLTouch probe on the bed when the nozzle was at X260 so probing the full 10x10 grid.

### Actual behavior

I cannot get the nozzle to move beyond X235

### Steps to Reproduce

1. In config.h I set MESH_INSET at 10 (ok for left, front and back) 
2. In config_adv.h I set MESH_ MAX_X as 260 (I enabled the others at their defaults)
3. I set M211 S0 to turn off endstops
4. I probed a 10x10 mesh
5. It achieved an 8x10 mesh
6. Setting MESH_MAX_X as X_BED_SIZE gives the same result. I have reverted to this.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6721504/Marlin.zip)

### Version of Marlin Firmware

2.0.9

### Printer model

Creality Ender 3 Pro

### Electronics

SKR 1.4

### Add-ons

BLTouch

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

badbod, 2021-06-30:
The mesh size should not go over 235, and if you want 100 probe points accessible it will need to be 10 smaller on the X and 5 smaller on the Y due to your probe offsets and available MAX_Y_POS / MAX_X_POS.

if you can move nozzle X to  260 then : 

configuration_adv.h set MESH_MAX_xxx  at defaults

configuration.h
#define X_BED_SIZE 235
#define Y_BED_SIZE 235
 
#define X_MAX_POS X_BED_SIZE + 25

You should now be able to move nozzle X to 260 (probe would be @ 225)

so 260 - 35 = 225 ,  235-225 = 10  so you now need mesh inset 10 with no margin

comment : 
//#define PROBING_MARGIN 10     no margin, or you reduce the probing area by 20 in X and Y axis before you start

and set
#define MESH_INSET  10

so as not to lose 20mm in both axis you could try MESH_INSET 5 and
 configuration_adv.h 
#define MESH_MAX_X X_BED_SIZE - 10
 
putting your probe 5mm into the bed on MIN_X , MIN_Y and as as far as it can reach on MAX_X  (225)

meltonpieman, 2021-06-30:
Thanks for the help. I have revised my config files and it works completely. I did implement your last advice and set MESH_INSET to 5. That left the probe very close on X_MIN so I increased it to 6 for safety.
At last my first full probe of the bed and 100 points with no need to infill.
It would be nice to find a resource that fully explained all the things in the config files for relatively simple users. However, with all the changes of names/actions between versions that would be difficult. Still what there is can be helpful and luckily there are knowledgable people like you who are prepared to help. Thanks again. Meltonpieman aka David

thisiskeithb, 2021-06-30:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-30 by thisiskeithb

badbod, 2021-07-03:
oops, sorry, did see this was not so much a bug report as a misunderstanding of the margins and offsets, please move to relevant area like discussion etc in case it may be helfpul to others and help prevent errant bug reports.