MarlinFirmware/Marlin #22239: Marlin\src\lcd\menu\menu_motion.cpp does not compile

BeePerNet, 2021-06-27:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

In file included from Marlin\src\lcd\menu\menu_motion.cpp:33:
Marlin\src\lcd\menu\menu_motion.cpp: In function 'void _menu_move_distance(AxisEnum, screenFunc_t, int8_t)':
Marlin\src\lcd\menu\menu_motion.cpp:176:34: error: 'MSG_MOVE_100MM' is not a member of 'Language_en'; did you mean 'MSG_MOVE_10MM'?
  176 |     if (LARGE_AREA_TEST) SUBMENU(MSG_MOVE_100MM, []{ _goto_manual_move(100); });

### Bug Timeline

From version 2.0.9.1

### Version of Marlin Firmware

2.0.9.1

BeePerNet, 2021-06-27:
To fix it, add PROGMEM Language_Str MSG_MOVE_100MM = _UxGT("Déplacer 100mm"); to all language

BeePerNet, 2021-06-27:
I'm doing it...

thisiskeithb, 2021-06-27:
See https://github.com/MarlinFirmware/Marlin/pull/22238

thisiskeithb, 2021-06-27:
Fix was merged. Closing.

thisiskeithb, 2021-06-28:
Patched in https://github.com/MarlinFirmware/Marlin/commit/15a50b19d79e9d1ab068719ceeeef0e2b4737039

closed 2021-06-28 by thisiskeithb