MarlinFirmware/Marlin #22240: [BUG] (bug summary)"Print Aborted" behavior inconsistent using IDEX printer

inspectionsbybob, 2021-06-27:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

The behavior for what happens when you "Abort" a print is inconsistent depending on which extruder is running at the time of the Abort.

If E0 Is running, the printer follows some script in MERLIN somewhere

if E1 is running, the print follows the "Print Finished" Gcode for the Model being printed

This is running on a JGMaker Artist-D IDEX printer

### Bug Timeline

Allways

### Expected behavior

I expect it to:

Do the same thing regardless of what Extruder is running

Do things Safely, not do any damage

PREFERED Action : Follow the "Print Finished" Gcode in the Models Gcode file

### Actual behavior

If E0 is running, the print stop, extruders goes UP a touch, then Home, and the bed comes all the way forward **at travel Speed or faster**, it appears to turn of heaters, the steppers stay energized.

If E1 is running, the print stops, the printer follows the "Print Completed" Gcode in the Model's file,  
which  for -MY- files I have it set to Raise the extruders, Home the extruders, Move the bed forward 75% SLOWLY/Safely, turn off Both Extruder heaters, turn of the bed heater, turn off the steppers... play a tone

Moving the model forward at travel speed or faster is Dangerous, blocks the controls in the front of -my- printer, and is rough on the printer.  

I

### Steps to Reproduce

Start a dual extruder print

Abort the print while extruder "X" is ruuning

resrart

Abort while the OTHER extruder is running

### Version of Marlin Firmware

2021.5.2

### Printer model

JGMaker Artist-D

### Electronics

stock

### Add-ons

Stock

### Your Slicer

IdeaMaker

### Additional information & file uploads

None

thinkyhead, 2021-07-30:
It sounds like it might be related to auto-parking or some other IDEX-specific voodoo. I know that @InsanityAutomation has been the most active in this topic, so possibly he has run into this at some point.

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

inspectionsbybob, 2021-07-30:
I think it still exists in the 2.0.7 branch as that is what the JGMaker
community build is based on...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

InsanityAutomation, 2021-07-30:
I know a couple other idex regressions crept in lately and have been in my sights to fix, just struggling on time. Auto park did get broken not long ago and @Roxy-3D was intending to take a look as it seemed she had time before I did. If she hasn't yet I'll see about testing one this weekend.

thinkyhead, 2021-07-31:
> I think it still exists in the 2.0.7 branch as that is what the JGMaker community build is based on...

If it is a confirmed bug in 2.0.7 then I will add a note about that in the "Known Issues" section of its release notes.

thinkyhead, 2021-07-31:
> Auto park did get broken not long ago…

Let's aim to add more comments and make this area of the code less brittle.

inspectionsbybob, 2022-04-01:
I havent done any Dual extruder work recently.
How the "end" behaved was different depending on which extruder was running at the time...  got my knuckles rapped more than once at an abort when the bed decided to move much faster than normal travel speed forward after the abort if it was E1...

closed 2022-06-11 by github-actions[bot]