MarlinFirmware/Marlin #22243: [BUG?] get warning: 'EEPROM' defined but not used [-Wunused-variable] with SKR2.0

GER-HerrS, 2021-06-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling with SKR 2.0 Rev. B (newest Marlin 2.0.9.1) I get the following warning:
libraries\EEPROM\src/EEPROM.h:250:20: warning: 'EEPROM' defined but not used [-Wunused-variable]

As it is my first SKR I don't know if that is an important warning.
I never got this warning using Arduino 2560 Boards like RAMPS or stock Creality Boards.
So I wonder what is the reason for this and what does this means to my compiled firmware (as it completed compiling)?
Is there a need to change some config or is this some bug like the LCD problem I had as well (https://github.com/MarlinFirmware/Marlin/issues/22239)?

### Expected behavior

I expected no warning because I didn't do changes concerning EEPROM.

### Actual behavior

I got the warning... .

### Steps to Reproduce

Use SKR 2.0 Rev. B board.

### Version of Marlin Firmware

2.0.1.9

### Printer model

Creality CR-10 V2

### Electronics

BTT SKR 2.0

slowbro, 2021-06-27:
I wasn't able to reproduce this. Can you upload your configuration files?

GER-HerrS, 2021-06-27:
Thanks for the quick response.
Here are the configs.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6722354/Config.zip)

slowbro, 2021-06-27:
The warning can be ignored. It was introduced with PR #22054.

GER-HerrS, 2021-06-27:
Thanks again for your quick response!
Then I can flash my firmware even with this warning. Wasn't sure if I it would brick my board or so one.

slowbro, 2021-06-27:
Yep, it will be fine. The EEPROM.h header was added for USB support but not all parts of it are used, hence the error.

GER-HerrS, 2021-06-27:
Then I close this issue now
Thanks again.

closed 2021-06-27 by GER-HerrS