MarlinFirmware/Marlin #22249: [FR] Prusa Laser Filament Sensor Integration

hsrmt, 2021-06-28:
### Is your feature request related to a problem? Please describe.

I want to use the laser sensor from the file "fsensor.cpp". Could someone give me a sense as to  how difficult this would be? 

https://store.metatechtr.com/filament-sensor
https://github.com/prusa3d/Prusa-Firmware 

Thank you so much :)

### Are you looking for hardware support?

https://github.com/prusa3d/PRUSA_Laser_filament_sensor

### Describe the feature you want

I want to be able to use the Prusa laser filament sensor

ellensp, 2021-06-28:
Dead product, no longer used by Prusa. Was to unreliable

hsrmt, 2021-06-28:
Thank you so much :) Is there another sensor that you could recommend to us?

thisiskeithb, 2021-07-01:
Prusa switched to an IR sensor with indirect filament presence detection using a ball bearing.

> Is there another sensor that you could recommend to us

[Prusa IR Sensor](https://shop.prusa3d.com/en/mk3mk3smk3s/921-ir-filament-sensor.html). There are also several third-party manufacturers, so search around if you can’t buy from Prusa.

This sensor works in Marlin without modification and I use it on a few builds here.

thisiskeithb, 2021-07-01:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/11693

closed 2021-07-01 by thisiskeithb