MarlinFirmware/Marlin #22250: [BUG] PAUSE_MESSAGE_PAUSING undefined

mjago, 2021-06-28:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Whilst running up the recent Elegoo Neptune printer on Marlin v2.0.9.1 I had an error due to undefined `PAUSE_MESSAGE_PAUSING`
https://github.com/MarlinFirmware/Marlin/blob/1dfa6cbc809d93a685c75f8b88ee3b9173aaeaa9/Marlin/src/lcd/extui/mks_ui/draw_pause_message.cpp#L32-L35
There is also a `PAUSE_MESSAGE_PARKING` that is defined elsewhere, which appears to be somewhat related. https://github.com/MarlinFirmware/Marlin/blob/3fa3be1f3103d70f4ae11ec6383984305dd95b1b/Marlin/src/feature/pause.h#L47-L50 The problem showed up on a MSK Robin Nano32 board with some pause/resume defines defined so will generally be hidden. I'm new to Marlin so I decided to just let you know, but if you need info please ask.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Elegoo Neptune 2

mjago, 2021-06-28:
I have the printer working perfectly with `v2.0.9.1` and `TFT_COLOR_UI` (which I really like and will keep). I'll investigate the LVGL problem for completeness but I suspect that PAUSE_MESSAGE_PAUSING == PAUSE_MESSAGE_PARKING.

thinkyhead, 2021-07-30:
Patched!

commit 14d40fb9 (2021-07-30): 🐛 Fix PAUSE_MESSAGE_PAUSING=>PARKING

Fixes #22250. Regression from #17460.
commit c7c56ac4 (2021-07-30): 🐛 Fix PAUSE_MESSAGE_PAUSING=>PARKING

Fixes #22250. Regression from #17460.
closed 2021-07-30 by thinkyhead