MarlinFirmware/Marlin #22251: [FR] Use same extruder for multiple tools

kurtis-potier-geofabrica, 2021-06-28:
### Is your feature request related to a problem? Please describe.

In RepRap I have the ability to define what extruder drives what tool using the M653 command in the config.g file. I can't find any similar setting in the marlin firmware. 

I've tried editing the pins file to define the same pins for multiple extruders as an attempt to create the same effect but it is creating issues. Strangely, this causes homing to fail, crashing the axis through the endstops. If I use the same configuration without multiple extruders with the same pins, homing works fine. 

### Describe the feature you want

What I want to do is have a common extruder stepper on my toolchanger that mates with whatever tool it picks up. Thus reducing weight of the tools and reducing the number of steppers/drivers I need. This feature would allow marlin to support many more tools than the amount of extruder drivers on a board. 

Thank you!

### Additional context

In RepRap I can define [M563](https://duet3d.dozuki.com/Wiki/Gcode#Section_M563_Define_or_remove_a_tool) in config.h. In this example all 4 tools are running off of drive 0:

```
; Tools
M563 P0 S"T0" D0 H1 							; Define tool 0
G10 P0 X0 Y40 Z-1							; Reset tool 0 axis offsets
G10 P0 R0 S0 								; Reset initial tool 0 active and standby temperatures to 0C

;M563 P1 S"T1" D0 H2 							; Define tool 1
;G10 P1 X-8.80 Y38.70 Z-4.99 				; Reset tool 1 axis offsets
;G10 P1 R0 S0 								; Reset initial tool 1 active and standby temperatures to 0C

M563 P2 S"T2" D0 H3 		; Define tool 2
G10 P2 X-41.5 Y77.0 Z-1.35					; Reset tool 2 axis offsets
G10 P2 R0 S0 								; Reset initial tool 2 active and standby temperatures to 0C

M563 P3 S"T3" D0 H4 	; Define tool 3
G10 P3 X-41.5 Y77.0 Z-1.60					; Reset tool 3 axis offsets
G10 P3 R0 S0 								; Reset initial tool 3 active and standby temperatures to 0C

```
I would like to do this with Marlin to make use of the more widely available boards and other features.

thinkyhead, 2021-07-30:
Marlin's tools are hard-coded at compile-time, and it is unlikely that this will be implemented for quite a while. Are you unable to get the desired behaviors using slicer parameters?

thinkyhead, 2021-07-30:
> common extruder stepper on my toolchanger that mates with whatever tool it picks up…

I see. On second thought, it should be possible to add a new sub-option to `SWITCHING_TOOLHEAD` so that the same E stepper is always used. The main changes would be in `stepper/indirection.h`.

thinkyhead, 2021-07-30:
Try this change in `stepper/indirection.h` and then define `SWITCHING_TOOLHEAD_SINGLE_STEPPER` in your configuration:
```diff
- #elif HAS_PRUSA_MMU2  // One multiplexed stepper driver
+ #elif EITHER(HAS_PRUSA_MMU2, SWITCHING_TOOLHEAD_SINGLE_STEPPER) // One multiplexed stepper driver
```