MarlinFirmware/Marlin #22254: Marlin not compiling mega i3  - MarlinUI::lcd_clicked during firmware.elf

ntangle, 2021-06-28:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Compiling Marlin 2.0.9 upwards (same for 2.0.9.1) fails at the end during the firmware.elf using the i3 mega example in the 2.0.9.x configs.

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\ammeter.cpp.o
Compiling .pio\build\mega2560\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\mega2560\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
Compiling .pio\build\mega2560\src\src\feature\controllerfan.cpp.o
Compiling .pio\build\mega2560\src\src\feature\e_parser.cpp.o
Compiling .pio\build\mega2560\src\src\feature\host_actions.cpp.o
Compiling .pio\build\mega2560\src\src\feature\pause.cpp.o
Compiling .pio\build\mega2560\src\src\feature\runout.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\G26.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\mbl\G29.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\mbl\M421.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G34_M422.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\M666.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M304.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M309.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M111.cpp.o
[... 149 lines cut ...]
### Bug Timeline

from 2.0.9

### Expected behavior

To finish compiling

### Actual behavior

Fails

### Steps to Reproduce

1.Download the 2.0.9.1 Zip, and the 2.0.9.1 configurations. Use the Anycubic i3 mega config/config.adv
2. Compile.
3. Fails!

### Version of Marlin Firmware

2.0.9.1

### Printer model

Anycubic Mega i3 S

ntangle, 2021-06-28:
Tried with 2.0.x bugfix - still same error

slowbro, 2021-06-28:
I can confirm this, same error using the i3 mega config. From the looks of it, it is not including/compiling the `menu/menu_*` files.

slowbro, 2021-06-29:
It looks like this is being caused by `G26_CLICK_CAN_CANCEL`; commenting it out like so..

```diff
diff --git a/Marlin/src/module/temperature.h b/Marlin/src/module/temperature.h
index 3a8c506a5d..6c191f01e3 100644
--- a/Marlin/src/module/temperature.h
+++ b/Marlin/src/module/temperature.h
@@ -175,7 +175,7 @@ enum ADCSensorState : char {
 #endif
 
 #if ENABLED(G26_MESH_VALIDATION) && EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
-  #define G26_CLICK_CAN_CANCEL 1
+  //#define G26_CLICK_CAN_CANCEL 1
 #endif
 
 // A temperature sensor
```

.. allows compilation to finish. Maybe missing an include of `marilinui.h` (where `MarlinUI::lcd_clicked` is defined) in one of the extui files..? Looking more, but the diff above will let you build for now.

slowbro, 2021-06-29:
Found it.

https://github.com/MarlinFirmware/Marlin/commit/a3e8bb99e472042c7db0b2d0be8d515ee715cf28#diff-3c8b6ef45c61c9cff05542db2f1b61eb8926eaf3e33f9b4d952323fdce5bddf8R226-R229

This is placed inside of a (file-global) `#if HAS_WIRED_LCD`, likely accidentally as the file isn't indented to show it (as usual) and it's kind of hidden. Moving it above that #if lets compilation happen..

```diff
diff --git a/Marlin/src/lcd/marlinui.cpp b/Marlin/src/lcd/marlinui.cpp
index d2e26afedf..ccaadf6c96 100644
--- a/Marlin/src/lcd/marlinui.cpp
+++ b/Marlin/src/lcd/marlinui.cpp
@@ -120,6 +120,11 @@ constexpr uint8_t epps = ENCODER_PULSES_PER_STEP;
   }
 #endif
 
+#if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
+  bool MarlinUI::lcd_clicked;
+#endif
+
+
 #if HAS_WIRED_LCD
 
 #if HAS_MARLINUI_U8GLIB
@@ -223,10 +228,6 @@ millis_t MarlinUI::next_button_update_ms; // = 0
 
 #endif
 
-#if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
-  bool MarlinUI::lcd_clicked;
-#endif
-
 #if HAS_LCD_MENU
   #include "menu/menu.h"
 
```

Jebdiah66, 2021-06-29:
Hi, maybe this would be a better solution:
[1.patch.txt](https://github.com/MarlinFirmware/Marlin/files/6733222/1.patch.txt)

A Printer like the Anycubic I3 Mega don't use HAS_LCD_MENU.

slowbro, 2021-06-29:
#22257 was merged, which will fix this.

@Jebdiah66 the `G26_CLICK_CAN_CANCEL` comment I made was me just being verbose; it wasn't the actual cause, just narrowing down the symptoms to an area that could be disabled for OP to build. HAS_LCD_MENU and EXTENSIBLE_UI are more-or-less mutually exclusive, so that `BOTH()` would never be satisfied. The issue was that `lcd_click` was extended to EXTUI (despite the name having 'lcd' in it) to 'simulate' clicks in marlinui & trigger associated actions.

peter-ch, 2021-06-30:
None of the fixes work. I tried commenting out "#define G26_CLICK_CAN_CANCEL 1" - didn't work, still can't compile. I tried replacing the EITHER with BOTH - didn't work.

peter-ch, 2021-06-30:
The error I get is: error: assignment of read-only variable 'MarlinUI::lcd_clicked' 
Is this related? How can I fix this fast?

thisiskeithb, 2021-06-30:
> #22257 was merged, which will fix this.

Confirmed. Default i3 mega config builds fine now.

> How can I fix this fast?

@peter-ch: Download a fresh copy of [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) since it has this fix.

PR #22257 (2021-06-29): Fix EXTUI "user click" (#22254)
### Description

[Concise Diff](22257/files?w=1)

Looks like in a3e8bb99e472042c7db0b2d0be8d515ee715cf28, the 'user_click' function was moved out of a `HAS_WIRED_LCD` conditional in the header file, but not in the cpp file.

### Requirements

Any EXTUI display with G26 Mesh validation enabled (as an example).

### Benefits

Fixes compilation issue for EXTUI.

### Configurations

Using example configs for "Anycubic i3 Mega"

### Related Issues

Resolves #22254
closed 2021-06-30 by thisiskeithb