MarlinFirmware/Marlin #22256: [BUG] AVR: DELAY_US(x>64) causes int overflow (affects M672, D100, minkill())

skruppy, 2021-06-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Platform: AVR / ATmega2560 with F_CPU 16e6

DELAY_US(x) has an internal type conversion from uint16 to uint8, which overflows if x > 64.

For example:
* While handling `D100` (requires `MARLIN_DEV_MODE` to be set in config) a loop calls 10,000 times `DELAY_US(1000UL)`.
* Which calls `DELAY_CYCLES(x)` with x (uint16) = [1,000 * (F_CPU / 1,000,000)](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/shared/Delay.h#L139) = 16,000
* Inside `DELAY_CYCLES` `x` gets divided by 4 and `__delay_4cycles(x)` will be [called](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/shared/Delay.h#L128).
* The argument of `__delay_4cycles` is an [uint8](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/shared/Delay.h#L103), which can't store the 4000.

As far as I can see this affects at least M672, D100 and minkill().

### Bug Timeline

Seems to be pretty old, but unnoticed

### Expected behavior

A proper watchdog induced board restart after 8 seconds.

### Actual behavior

After roughly a Seconds, the message ''FAILURE: Watchdog did not trigger board reset.'' appears.

### Steps to Reproduce

1. Send ''D100''

### Version of Marlin Firmware

2.0.8.1

slowbro, 2021-06-29:
I spent some time looking up ASM inline, googlin' stuff, and reading of the AVR instruction set. I figured it would be a fun learning opportunity! I was kind of wrong, lmao, but I *was* able to come up with this...

```cpp
  FORCE_INLINE static void __delay_4cycles(uint16_t cy) {
    uint16_t dummy;

    __asm__ __volatile__(
      A("mov %A[reg], %A[cnt]")
      A("mov %B[reg], %B[cnt]")
      L("1")
      A("sbiw %A[reg],1")
      A("brne 1b")
      : [reg] "=&w" (dummy)  // high+low reg for uint16
      : [cnt] "w" (cy)       // input: number of loops                                                                                                                                                                                         
      : "cc"                 // clobbers:
    );
  }
```

That is still 4 cycles since `sbiw` (subtract immediate value from word) is 2 cycles, vs `sub` at only 1. From my understanding, `=&w` sets up a write-only output register, and the `%A`/`%B` puts the LSB/MSB in subsequent (increasing) registers.

I'm not sure entirely how portable, or indeed functional that is.. I tested it on an Arduino nano I had at my desk, and also got Marlin to compile with it for an atmega2560. The nano delayed about 10 seconds with 10k DELAY_US(1000), maybe that's right? You mentioned 8 sec.

Anyways, there's that!

skruppy, 2021-06-30:
Thank you @slowbro, your suggestion was a great starting point.

> I'm not sure entirely how portable, or indeed functional that is.. I tested it on an Arduino nano I had at my desk, and also got Marlin to compile with it for an atmega2560.

I was only able to tested PR #22268 with an ATmega2560. 

> You mentioned 8 sec.

sorry. I ment 4 seconds, the default watchdog timeout, the time after witch the board should reset after sending `D100`.

PR #22268 (2021-07-09): Fix int overflow in DELAY_US on AVR platform (closes #22256)
### Description

Complete rewrite of `DELAY_CYCLES(x)` for AVR platform. This in first place fixes an integer overflow (described in #22256) and uses a <del>GCC intrinsic for the compile time branch</del>16 bit version of the delay loop and an simpler inline ASM function with less overhead for the runtime branch.

### Requirements

  * AVR Platform (e.g. ATmega2560)

### Benefits
Fixes integer overflow in `DELAY_US(x)` on AVR platforms for x > 64

### Configurations
tested in `MARLIN_DEV_MODE` with the following code in `gcode_d.cpp` [... cut]
closed 2021-07-09 by skruppy